bdring/FluidNC #239: Problems Compiling Firmware

mhtngrbfe, 2022-01-10:
Have you read the [wiki on how to compile] (https://github.com/bdring/FluidNC/wiki/Compiling-the-firmware)?
Yes
Are you using upstream FluidNC or Devt?
FluidNC
Have you made any edits or configuration changes (please list) to the firmware?
Not
It is possible to add an option for:
https://github.com/FYSETC/FYSETC-E4

MitchBradley, 2022-01-10:
You can automatically convert any machine.h file that you like with the automated converter.  Here is a related conversion: https://github.com/bdring/Grbl_Esp32/issues/1046

bdring, 2022-01-10:
There is also a 3 axis corexy example here.

https://github.com/bdring/FluidNC/blob/main/example_configs/fysetc_corexy.yaml

closed 2022-01-10 by bdring

mhtngrbfe, 2022-01-10:
Hi!
Thanks a lot! The board hasn't left China yet.
First I studied the question, then I ordered it. After that I asked you a
question. I will work on how he arrives.
I want to put on a portal with two Y's. And use the parking from the
drivers.
Thanks again for responding so quickly!
I am sorry for my English! :) Google!
I live in Ukraine.

-- 
С уважением,
Фертенко Александр
Ведущий инженер по внедрению
новой техники и технологий
ТОВ "ИНТЕРТЕХГРУП"
моб.: +38 093 393 07 47
моб.: +38 050 351 52 09
моб.: +38 097 0 120 140
***@***.***