bdring/FluidNC #240: Support ARM linux in install scripts.

easytarget, 2022-01-10:
My controller is permanently connected to a Raspberry PI, and I wanted to install and upgrade from that since it was already connected. 

So the specific, low priority, 'feature request' is for 32-bit armv7 install support in the release tooling.
- adding arm6 support would also cover the Pi1, Pi2 and v1 Pi Zero, but is probably less relevant
- 64bit arm support is not really an issue at present, but an official 64bit RasPI OS will happen eventually

With the install scripting you obviously have allowed for adding other architectures going forward, Until then I wanted to document how I got it working somewhere more public than Discord. 

----------

First, install esptool for the arm architecture; the quickest commandline way to get this is via python/pypi
```$ pip install esptool```

Then edit `tools.sh` in the unpacked zip folder and replace line 3 with:
```EsptoolPath=esptool.py```

Then follow instructions as normal.

Caveats:
This was done using a fully updated Bullseye release with Python3 as the default.
- afik this wil also work just fine on the older Stretch and Buster releases, and should also work for Python2. But I have no systems to test this on so ymmv.
- There is a version of esptool available via apt; `sudo apt install esptool` followed by `EsptoolPath=esptool` should work too; and may even be better advice. But I have not tested that, the pypi version of esptool is generally much newer.

MitchBradley, 2022-01-10:
We will accept a PR to add Raspberry Pi support to the release scripting if you author it and agree to support if/when people have problems.

easytarget, 2022-02-04:
As you identify, supporting various install scripts with binary blobs for differing OS's is not fun, and not something I'd be prepared to take on.

However; as an alternative, a generic installer that uses python directly with esptool.py, and avoids distributing any executables seems like a much better long-term solution.

I've got this ready, tested and working on a Pi4+Bullseye, Pi3a+Buster and my Fedora35 workstation. I am just fixing the readme and will drop a PR against DevT asap.

bdring, 2022-02-04:
Distributing a python script to Windows is a pain.

MitchBradley, 2022-02-04:
We are amenable to using the python + *.py technique on Linux and MacOS.  The precompiled executable method works pretty well on Windows, but is trouble-prone on other OSs.

easytarget, 2022-02-15:
#296 closes this nicely.

closed 2022-02-15 by easytarget