bdring/FluidNC #248: Degraded Axis Squaring

sluebbert, 2022-01-15:
I believe the behavior for homing a squared axis has changed from Grbl_ESP32 to FluidNC. I didn't change any hardware or wiring during the swap. In Grbl_ESP32 the squaring worked fine. Both motors would bring the axis to the origin, both sides would back off and then kiss the end stops once and be done.
In FluidNC the behavior is both motors bringing the axis to the origin. Assuming I have an axis that is slightly skewed because I moved the axis away manually, one side will beat the other to trigger the end stop. This I expect. What then happens though is the entire axis does the back off like normal, but the other side only moves slightly towards its end stop. It goes several millimeters and then gives up giving me the message of:
`Homing fail. Could not find limit switch within search distances. Try increasing max travel, decreasing pull-off distance, or check wiring.`
So I home again, and again, and so on until the skewed axis catches back up and hits its end stop.
Unfortunately this dance (the nudge towards the end stop, the back off on a successful hit) from the independent sides can go on for a while. The axis just skews slightly to the left, then to the right, and so on if I keep clicking home over and over. To get around this I have to help the homing manually by waiting until both sides get close enough, and then nudging the sides to be as straight as possible, and then re-home.
If the ganged axis is straight, and stays straight, homing always works fine.

It just seems like everything would work fine if the skewed behind side would just keep moving towards origin to find the end stop instead of failing after several millimeters.

I currently have two independent motors, two independent end stops that are wired in series using a single IO pin to make up the axis.

Am I missing something in my configuration? Do I need to alter the physical setup or wiring of my end stops?
Maybe the WIKI info on stress free squaring [here](https://github.com/bdring/FluidNC/wiki/FluidNC-Homing#ganged-motors-with-one-switch-input) covers it all? Its just a little ambiguous to me. I'm just looking for some guidance before I start rewiring. Would splitting my two end stops to be two separate IO pins instead of being in series mimic the old behavior I'm aiming for?

What version of the firmware are you using?
3.3.0

Insert boot messages below.
```
[MSG:INFO: FluidNC v3.3.0]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key spinup_ms]
[MSG:WARN: Ignored key spindown_ms]
[MSG:WARN: Ignored key speeds]
[MSG:INFO: Machine TMC2209 XYY Laser Controller]
[MSG:INFO: Board TMC2209 4 Axis]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:3us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.25]
[MSG:INFO: Axis X (0.000,1030.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Trinamic Stepper Uart Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:INFO:     tmc_2209 Step:gpio.26 Dir:gpio.27 Disable:NO_PIN Addr:0 R:0.110]
[MSG:INFO:     driver test passed]
[MSG:INFO:     All Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,1010.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 Step:gpio.33 Dir:gpio.32 Disable:NO_PIN Addr:1 R:0.110]
[MSG:INFO:     driver test passed]
[MSG:INFO:     All Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_2209 Step:gpio.2 Dir:gpio.14 Disable:NO_PIN Addr:2 R:0.110]
[MSG:INFO:     driver test passed]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Spindle Ena:NO_PIN Out:gpio.4 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Connecting to STA SSID:****]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.146]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.3 [FluidNC v3.3.0 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
```

config.yaml:
```yml
name: TMC2209 XYY Laser Controller
board: TMC2209 4 Axis

kinematics:
  Cartesian:

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 3
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.25
  x:
    steps_per_mm: 80.1
    max_rate_mm_per_min: 9000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 1030.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      feed_mm_per_min: 400
      seek_mm_per_min: 2000
    motor0:
      limit_all_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 2.000
      tmc_2209:
        direction_pin: gpio.27
        step_pin: gpio.26
        r_sense_ohms: 0.110
        addr: 0
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
        uart:
          txd_pin: gpio.22
          rxd_pin: gpio.21
          baud: 115200
          mode: 8N1
    motor1:
      null_motor:
  y:
    steps_per_mm: 80.1
    max_rate_mm_per_min: 9000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 1010.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      feed_mm_per_min: 400
      seek_mm_per_min: 2000
    motor0:
      limit_all_pin: gpio.34:high
      hard_limits: true
      pulloff_mm: 2.000
      tmc_2209:
        direction_pin: gpio.32
        step_pin: gpio.33
        r_sense_ohms: 0.110
        addr: 1
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false
    motor1:
      pulloff_mm: 2.000
      hard_limits: true
      tmc_2209:
        direction_pin: gpio.14
        step_pin: gpio.2
        r_sense_ohms: 0.110
        addr: 2
        run_amps: 0.250
        hold_amps: 0.125
        microsteps: 16
        stallguard: 16
        stallguard_debug: false
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: false

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:

coolant:
  delay_ms: 1000.000

probe:
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speeds: 0=0.0% 1000=100.0%
  output_pin: gpio.4
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 5000
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false 
```

Much of the config file was auto generated by the tool/bot from the grbl esp32 repo, including the laser section keys it is saying is ignored.

Northernboy, 2022-01-15:
For Axis Squaring to work in FluidNC you need a separate switch and input for each of your Y Axis. Your present setup doesn’t do axis squaring as you have noticed.

sluebbert, 2022-01-16:
Alright that clarified my suspicions and finally gave me a good reason to fix some end stop wires that were annoyingly short. It works as expected now, thanks!

closed 2022-01-16 by sluebbert