bdring/FluidNC #251: Trouble uploading firmware

SawDust58, 2022-01-16:
I am having trouble installing FluidNC firmware. I downloaded and extracted v3.3.0 zip file onto my win10 64bit computer. I tried installing using win64\install-fs.bat but there is a problem finding esptool:

win64\esptool.exe --chip esp32 --baud 921600 dump_mem 0x3ff5a018 4 SecurityFuses.bin
The system cannot find the path specified.
esptool failed

I have used esptool in the past to successfully flash my esp32 and esp8266 boards.

Then I tried uploading directly using platformio within vscode. I opened FLUIDNC-MAIN folder then hit upload button:

> Executing task: C:\Users\Jeff\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing wifi (platform: espressif32; board: esp32dev; framework: arduino)
SCons import failed. Unable to find engine files in:
  C:\Users\Jeff\.platformio\packages\tool-scons\..
  C:\Users\Jeff\.platformio\packages\tool-scons\scons-local-4.1.0
Traceback (most recent call last):
  File "C:\Users\Jeff\.platformio\packages\tool-scons\scons.py", line 87, in <module>
    import SCons.Script
ModuleNotFoundError: No module named 'SCons'
[FAILED] Took 0.20 seconds 

Environment    Status    Duration
-------------  --------  ------------
wifi           FAILED    00:00:00.197
=== 1 failed, 0 succeeded in 00:00:00.197 
The terminal process "C:\Users\Jeff\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

I did try 'pip install -U scons' and scons installed OK but this did not fix platformio upload problem.
Do I need to edit vscode settings to find these files? Any help would be appreciated.

Note: previously I had Grbl_Esp32 operating on the same ESP32, but I thought I should switch to FluidNC for latest developments.

MitchBradley, 2022-01-16:
"I tried installing using win64/install-fs.bat".  What exactly does that mean?  Did you double-click that item?  Before or after unzipping?  Or did you type some command?  If so, exactly what did you type and from what command interpreter inside what directory?

SawDust58, 2022-01-16:
Thanks Mitch for the prompt reply.
I extracted FluidNC-main zip file to a new folder where I collect my other platformio projects. I then drilled into this folder to FluidNC-main\install_scripts\win64 folder and doubled clicked on install-fs.bat file name since this is the initial install (per HOWTO-INSTALL.txt instructions). A command window opens (cmd.exe) and I immediately get the 'esptool failed' message I showed above. When I open 'install-fs.bat' in Notepad I see the command line 'set EsptoolPath=win64\esptool.exe'. The win64 folder inside the install_scripts folder does not contain esptool.exe file, and a search on my C: drive for win64\esptool.exe returns zero matches (when I search C: drive for esptool.exe I get a match at 'C:\Users\Jeff\.platformio\packages\tool-esptool\esptool.exe'). I trust the installer scripts are working for other users, however I don't know how to make it happy on my computer.

MitchBradley, 2022-01-16:
Oh!  I think I see the problem.  You used a source zip instead of a release package.  You need to go to [the releases page](https://github.com/bdring/FluidNC/releases) and get a release bundle zip file. In your case it would be fluidnc-v3.3.0-win64.zip .  The instructions in HOWTO-INSTALL.txt are correct for the release bundle file layout, not the source tree.  The source tree does not contain the various precompiled binaries that are present in the release bundles.

By the way, what kind of bit did you use when you "drilled into this folder"?

MitchBradley, 2022-01-16:
Regarding the difficulty with platformio:

In my setup, the platformio upload starts as follows:

> Executing task: C:\Users\wmb\.platformio\penv\Scripts\platformio.exe run --target upload <

Notice that there is a \ between wmb and .platformio. That backslash is missing in all of the paths that you mentioned in your report.  Conventionally, .platformio is a subdirectory of your user directory.  I have no idea why it looks different in your setup, but if it is trying to use wrong pathnames that should have a backslash in them, the files will not be found.

I am not sure about this next part, but I think that the problem is not in platformio itself, but instead in either the systemwide installation setup - perhaps a path-related environment variable,- or in the vscode to platformio integration setup.  The reason I think that is because the problem is apparent in the initial "Executing task" message, and that message comes from the vscode integration, not from inside platformio.  So the root of the problem has to be either in the vscode/platformio setup or even further out.

SawDust58, 2022-01-16:
> You used a source zip instead of a release package.
Mitch,
Yes, that is exactly what I did, and I now see the wiki told me to go to Releases. I downloaded latest release package and FluidNC installed flawlessly, very nicely done! 

Regarding the bit I used to drill into Folders, it was a virtual bit that cuts perfectly or not at all depending on how you look at it. I am considering building a CNC router and am investigating what software and electronics are required and what products I want to use (no hardware yet other than a spare ESP32).

Thanks for your help and patience supporting new users, I will close this Help issue.

closed 2022-01-16 by SawDust58