bdring/FluidNC #256: Linux fluidterm.sh is calling pip install

jeffeb3, 2022-01-18:
**Please describe the feature you would like implemented**
I don't like the fluidterm.sh calling pip install without warning. What I think would be better is either a check "You don't have pyserial or xmodem installed. Exiting" or a prompt to install it "You don't have pyserial installed, would you like to install it?".

I edited the install script to not call fluidterm.sh, and then I made a virtualenv before calling fluidterm.sh. 

**Why do you think this would improve FluidNC?**
It isn't perfect, but I manage my pip installs very carefully, and installing these from a script is going to mess with my host's packages.

**What do you need the feature for?**
Just to avoid changing the global packages installed on my system.

**Do you know of other users who need this feature?**
Any Linux users would appreciate it, I think. I don't know enough about windows or Mac installs to say for them.

I am bringing this up, in case it hasn't occurred to anyone that it is a problem. I am not the most common user, so I understand if I just have to apply my paranoia to work around these issues myself. What I'm saying is that I am totally fine with you closing this issue if you've considered it, and disagree with me.

MitchBradley, 2022-01-18:
Congratulations!  You are now the maintainer of the Linux install script.  Please issue a PR for any changes that you wish to make, and watch this list and the Discord for anybody who is having problems.

jeffeb3, 2022-01-21:
Well. That would be fun. I am a little behind on my backlog of hobby responsibilities. But I will put this on the list.

easytarget, 2022-02-05:
For reference: #272 
- we could change the instructions (`HOWTO-INSTALL.txt`) so they guide the user through installing using a venv, it's really not that difficult. I could add that to the PR. (text adapted from the first part of similar instructions I wrote [here](https://github.com/easytarget/pi-overwatch/blob/main/docs/INSTALL.md))

MitchBradley, 2022-02-05:
The goal was "download/unzip the installer and click on something".  In my experience with naive users, "it's really not that difficult" is not the way things work out.  With desktop Linux, you can usually assume that if they are using Linux, they have already come to grips with basic command line usage and package installation.  With Raspberry Pi, there seem to be a lot of people that haven't the slightest clue about those basic tasks.

There might not be a solution.  If Linux was easy to use it wouldn't be Linux.

easytarget, 2022-02-05:
Yep, RasPi support is a nightmare; I'm supporting people on it elsewhere.

I've got a pretty vanilla pi with a desktop and a esp32 devboard plugged in as a test target, I'll put my 'noob' hat on and see what happens when I do this by clicking icons.. It might work.. or might not. But I get where you are coming from, so I'll see if anything sucks, or if install guide changes seem necessary.

easytarget, 2022-02-06:
I did some testing on the user experience with install on Linux desktops; weirdly, the PI performed best.
- With a Pi+Desktop and filemanager (lxde-filemanger) you can select the zip, extract it and then doubleclick 'xxx-install.sh', you get asked what to do, and you can select 'run in terminal' at that point and everything just works.
- On anything with the default Nautilus file manager (Fedora, Ubuntu, etc) you are fully and systematically blocked from executing anything by clicking on it in any way; no exceptions. Which is good for protecting users, but not so great for anybody trying to distribute code.
- Using the inbuilt 'archive manager's to view and doubleclick will always fail; they extract to `/tmp`, which is mounted `--no-executable` on any modern machine.

The HOWTO-INSTALL.TXT would benefit from a bit of a restructure/rewrite to make things simpler for all audiences, especially novices. I think it should say that command line install is the only supported method. I'll look at that tomorrow while doing some other changes on #272 .
 
ps: /rant/ Looking at the way it is going I'm pretty certain the ultimate fate of Nautilus is to drop any pretence at showing the actual file structure; and it will simply become a beautifully styled search box and results list.