bdring/FluidNC #259: direction_pin: gpio.22

DEST1981, 2022-01-25:
Hi.
Release v3.3.1 

I set in config.yaml direction_pin: gpio.22

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 400
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.35:low
      standard_stepper:
        direction_pin: gpio.22:high
        step_pin: gpio.21
    motor1:
      null_motor:
But it is not work. The direction_pin: gpio.22 not change status low\hi if i chande direction.

MitchBradley, 2022-01-25:
How are you testing the state of the direction pin? What is your hardware? What is the rest of your config? What are the startup messages on the console?

It is impossible to solve a problem with the very small amount of information that you provided.

bdring, 2022-01-25:
Please post your boot messages.

https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages

DEST1981, 2022-01-25:
> How are you testing the state of the direction pin? 
>
Аn oscilloscope
>What is your hardware? 
>
Driver TB6600, main plate DIY (testing in GRBL ESP32)
>What is the rest of your config? 
>
name: "ESP32 Dev Controller V4"
board: "ESP32 Dev Controller V4"

stepping:
  engine: RMT
  idle_ms: 0
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: gpio.13:high
  
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 400
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.35:low
      standard_stepper:
        direction_pin: gpio.22
        step_pin: gpio.21
    motor1:
      null_motor:

  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 4000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 400
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false

    motor0:
      limit_all_pin: gpio.34:low
      standard_stepper:
        direction_pin: gpio.17:low
        step_pin: gpio.16
    motor1:
      null_motor:

  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 220
    homing:
      cycle: 1
      mpos_mm: 10
      positive_direction: true

    motor0:
      limit_all_pin: gpio.39:low
      standard_stepper:
        direction_pin: gpio.12
        step_pin: gpio.14
    motor1:
      null_motor:

    
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin:  NO_PIN

        
probe:
  pin: gpio.32:low:pu
  check_mode_start: false

PWM:
  pwm_hz: 35000
  output_pin: gpio.2
  enable_pin: gpio.15
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 2000=25% 6000=50% 10000=100%

>What are the startup messages on the console?
> 
[MSG:INFO: Restarting]
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Grbl 3.3 [FluidNC v3.3.1 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

I changed the direction_pin: gpio.22 to gpio.4 and everything works correctly. Hardware changed too.

bdring, 2022-01-25:
You skipped all the meaningful stuff in the boot messages. gpio.22 should work. Try to use gpio.22 for something like mist coolant to see if it is a hardware issue.

MitchBradley, 2022-01-25:
Maybe there is a short on the GPIO 22 wiring.

MitchBradley, 2022-01-26:
If you want us to engage with this problem you will have to give us lots of information, instead of doling it out one drop at a time.  For example
* we need all the boot messages
* "oscope" is a weak answer to "how".  We need to know where you measured (right at the GPIO pin, or elsewhere on the board).  We need to know the commands that you used to change the direction pin value.  Did you try measuring with the ESP32 unplugged from the board?
* Information like "it worked fine with Grbl_Esp32" should have been provided at the outset, not after a lot of prodding
* Since it worked with Grbl_Esp32, it would have been helpful if you had attached the machine.h file.  If the  FluidNC config file was auto-generated, a pointer to the issue that generated it would suffice.

We would like to help, but we are very busy.  When you give us too little information, it wastes everybody's time because the search space of possible problems is very large.  Asking questions over and over is very time consuming, and also very frustrating.

DEST1981, 2022-01-26:
Dear Mitch. English is not my native language.
If you don't have time and you're also very frustrating, just say fuck off.
Now for the problem.
I had a working CNC on ESP32-GRBL. I recently saw that you are developing a new ESP32 project for a machine tool. I used FluidNC, I ported the settings from machine.h to config.yaml. Then the X axis stopped working. I connected the my analog oscilloscope to the ESP32 pin and noticed that when the direction is changed, the state of the pin does not change. Then I removed the board, cut the track and connected it to the gpio.4 pin, and everything worked fine. 

machine.h
>#ifdef N_AXIS
        #undef N_AXIS
#endif
#define N_AXIS 3
#define X_STEP_PIN              GPIO_NUM_21
#define X_DIRECTION_PIN         GPIO_NUM_22
#define Y_STEP_PIN              GPIO_NUM_16
#define Y_DIRECTION_PIN         GPIO_NUM_17
#define Z_STEP_PIN              GPIO_NUM_14
#define Z_DIRECTION_PIN         GPIO_NUM_12
//#define A_STEP_PIN              GPIO_NUM_26
//#define A_DIRECTION_PIN         GPIO_NUM_27
#define STEPPERS_DISABLE_PIN    GPIO_NUM_13
#define SPINDLE_TYPE            SpindleType::PWM
#define SPINDLE_OUTPUT_PIN      GPIO_NUM_2   	// labeled SpinPWM
#define SPINDLE_ENABLE_PIN      GPIO_NUM_15  	// labeled SpinEnbl
#define DEFAULT_SPINDLE_FREQ 	35000 // $33 Hz (extended set)
#define X_LIMIT_PIN             GPIO_NUM_35
#define Y_LIMIT_PIN             GPIO_NUM_34
#define Z_LIMIT_PIN             GPIO_NUM_39
//#define A_LIMIT_PIN     		GPIO_NUM_36
#define PROBE_PIN               GPIO_NUM_32
#define COOLANT_MIST_PIN        GPIO_NUM_33  
>

console startup
>[MSG:INFO: Restarting]
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
[MSG:INFO: FluidNC v3.3.1]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config3ax.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (10.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.21 Dir:gpio.22 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.16 Dir:gpio.17:low Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-210.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.15 Out:gpio.2 Dir:NO_PIN Freq:35000Hz Res:11bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:DEST]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.89.25]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
>

commands to change the direction pin value. 
>$J=G91 G21 F1000 X10
$J=G91 G21 F1000 X-10
>

Еhere is one more BUG. When I set the idle_ms: 250 and quickly click on the move buttons in the WebUI or program G-Code sender, the motor stopped. I continued to click the machine changed the coordinates but the motors did not move. With a idle_ms: 0 there is no problem. 

in commands windows
><Idle|MPos:9.000,-23.100,0.000|FS:0,0>
$J=G91 G21 F1000 Y-1
$J=G91 G21 F1000 Y-1
$J=G91 G21 F1000 Y-1
ok
<Jog|MPos:9.000,-23.620,0.000|FS:257,0>
ok
<Jog|MPos:9.000,-24.140,0.000|FS:840,0>
ok
<Jog|MPos:9.000,-25.160,0.000|FS:840,0>
$J=G91 G21 F1000 Y-1
$J=G91 G21 F1000 Y-1
$J=G91 G21 F1000 Y-1
$J=G91 G21 F1000 Y-1
$J=G91 G21 F1000 Y-1
ok
<Jog|MPos:9.000,-26.170,0.000|FS:840,0>
ok
<Jog|MPos:9.000,-27.090,0.000|FS:600,0>
ok
<Jog|MPos:9.000,-28.080,0.000|FS:384,0>
ok
<Jog|MPos:9.000,-28.810,0.000|FS:500,0>
<Jog|MPos:9.000,-29.750,0.000|FS:139,0|WCO:4.800,-3.000,-35.000>
ok
$J=G91 G21 F1000 Y-1
$J=G91 G21 F1000 Y-1
$J=G91 G21 F1000 Y-1
<Jog|MPos:9.000,-30.100,0.000|FS:600,0|Ov:100,100,100>
ok
<Jog|MPos:9.000,-30.870,0.000|FS:653,0>
ok
<Jog|MPos:9.000,-32.090,0.000|FS:600,0>
ok
<Jog|MPos:9.000,-32.980,0.000|FS:314,0>
<Idle|MPos:9.000,-34.090,0.000|FS:0,0>
>

If you need more information just tell me.

DEST1981, 2022-01-26:
My Hardware:
[image]

DEST1981, 2022-01-26:
[config_gpio4.txt](https://github.com/bdring/FluidNC/files/7943497/config_gpio4.txt)
console
>rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
[MSG:INFO: FluidNC v3.3.1]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config3ax.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (10.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.21 Dir:gpio.4 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.16 Dir:gpio.17:low Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-210.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.15 Out:gpio.2 Dir:NO_PIN Freq:35000Hz Res:11bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:DEST]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.89.25]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
>

[config_gpio22.txt](https://github.com/bdring/FluidNC/files/7943515/config_gpio22.txt)
console
>rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
[MSG:INFO: FluidNC v3.3.1]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config3ax.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (10.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.21 Dir:gpio.22 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.16 Dir:gpio.17:low Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-210.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.15 Out:gpio.2 Dir:NO_PIN Freq:35000Hz Res:11bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:DEST]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.89.25]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
>

DEST1981, 2022-01-26:
mist_pin:  gpio.22 work correctly. The state of the pin change hi/low.

bdring, 2022-01-26:
It appears that something has changed. Your second post of the gpio.22 boot messages version no longer shows multiple instances of Motor0.

- Does the gpio.22 version now work?
- When it did not work, what was the behavior of the motor. Did it not move at all or only move in one direction regardless of the direction specified.

DEST1981, 2022-01-26:
Motor move only in one direction regardless of the direction specified.

bdring, 2022-01-26:
- Can you explain the change in the boot messages?
- Is the gpio.22 version working now?

Please note there are 2 questions here and answer both in your next reply

DEST1981, 2022-01-26:
[MSG:INFO: Restarting]
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.3.1]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config3ax.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine ESP32 Dev Controller V4]
[MSG:INFO: Board ESP32 Dev Controller V4]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13]
[MSG:INFO: Axis X (10.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.21 Dir:gpio.22 Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (10.000,410.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.16 Dir:gpio.17:low Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-210.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.14 Dir:gpio.12 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.15 Out:gpio.2 Dir:NO_PIN Freq:35000Hz Res:11bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: Connecting to STA SSID:DEST]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.89.25]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

DEST1981, 2022-01-26:
>Can you explain the change in the boot messages?
>

I do not know

MitchBradley, 2022-01-26:
Can you try measuring the signal on GPIO 22 when the ESP32 is not plugged in to the board?

DEST1981, 2022-01-26:
Full flashing with flash reset, solved the problem with GPIO 22. I don’t know what the problem was. Sorry.

DEST1981, 2022-01-26:
But this problem is relevant. 
>Еhere is one more BUG. When I set the idle_ms: 250 and quickly click on the move buttons in the WebUI or program G-Code sender, the motor stopped. I continued to click the machine changed the coordinates but the motors did not move. With a idle_ms: 0 there is no problem.
>

MitchBradley, 2022-01-26:
I will look into that problem.  There was a change to the idle code not so long ago.  Maybe it broke something.

bdring, 2022-01-27:
You could try adding a duration to the **disable_delay_us:** config item. The default is 0

Some of the cheap TB6600 driver boxes are notoriously slow to respond to signals. 

```yaml
stepping:
  engine: I2S_static
  idle_ms: 250
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0
```
https://github.com/bdring/FluidNC/wiki/FluidNC-Motor-Setup#stepping

bdring, 2022-01-27:
Added a [card](https://github.com/bdring/FluidNC/projects/1#card-76787836) to the project tracker

PR #261 (2022-02-06): Cancel pending stepper disable on wake_up
Fixes #259 

Here is what I think was happening.  With /stepping/idle_ms=250, issue a series of short jogs using e.g. "$J=G91 G21 F1000 Y-1" with very brief pauses between them.  Depending on the timing, it is possible that one of the jogs completes and the system sets the idle timer.  Before the timer expires, another jog command comes in and calls Stepper::wake_up().  That turns on the enable (it was already on) but does not cancel the timer.  The motion starts, but before it is complete the timer expires and turns off the enable, thus stopping the m [... cut]
closed 2022-02-06 by MitchBradley