bdring/FluidNC #262: Unable to use 4th Axis as per Gcode

ASHAPURAAHM, 2022-01-28:
When we run below command

G3X32.5000Y54.0000R20.0000A180.0

x and y axis move to X32.5000Y54.0000 after completion of this A axis move 180 step

When we use MACH3 using same G code A axis also move simultaneously i.e XYA move simultaneously

link of video https://youtu.be/C0i3FtPryWY (done using [MACH3)]

fluid NC version 3.2.7
webui version 2.1b72
below is complete Gcode

G21
G17
G90
F2000.0000
M3
G0Z10.0000
G1X111.0000Y49.0000A0.0
G1Z-40.0000F2000.0000
G1X111.0000Y62.0000A0.0
G0Z10.0000
G0X72.5000Y54.0000A0.0
G1Z-50.0000F2000.0000
G3X32.5000Y54.0000R20.0000A180.0
G3X72.5000Y54.0000R20.0000A360.0
G0Z10.0000F1500
G0X0.0000Y0.0000A0.0F2500
M5

Below is configuration file

ame: "ESP32 V4"
board: "ESP32"
meta: "N_ver"

stepping:
engine: RMT
idle_ms: 250
dir_delay_us: 1
pulse_us: 2
disable_delay_us: 0

axes:
shared_stepper_disable_pin: NO_PIN

x:
steps_per_mm: 40
max_rate_mm_per_min: 2000.000
acceleration_mm_per_sec2: 50.000
max_travel_mm: 900.000
homing:
cycle: 2
mpos_mm: -3.000
feed_mm_per_min: 100.000
positive_direction: false

motor0:
  limit_all_pin: gpio.34:low
  hard_limits: false
  pulloff_mm: 3.000
  stepstick:
    direction_pin: gpio.2
    step_pin: gpio.0
y:
steps_per_mm: 40
max_rate_mm_per_min: 2000
acceleration_mm_per_sec2: 50
max_travel_mm: 500
homing:
cycle: 2
mpos_mm: -3
positive_direction: false

motor0:
  limit_all_pin: gpio.35:low
  hard_limits: false
  pulloff_mm: 3
  stepstick:
    direction_pin: gpio.15
    step_pin: gpio.26
z:
steps_per_mm: 200
max_rate_mm_per_min: 2000
acceleration_mm_per_sec2: 50
max_travel_mm: 75
homing:
cycle: 1
mpos_mm: 3
positive_direction: true

motor0:
  limit_all_pin: gpio.36:low
  hard_limits: false
  pulloff_mm: 3
  stepstick:
    direction_pin: gpio.33
    step_pin: gpio.27
a:
steps_per_mm: 89
max_rate_mm_per_min: 200
acceleration_mm_per_sec2: 25
max_travel_mm: 365

motor0:
  stepstick:
    direction_pin: gpio.14
    step_pin: gpio.12
spi:
miso_pin: gpio.19
mosi_pin: gpio.23
sck_pin: gpio.18

sdcard:
cs_pin: gpio.5
card_detect_pin: NO_PIN

coolant:
flood_pin: gpio.21

probe:
pin: gpio.32:low:pu

user_outputs:
digital0_pin: gpio.25

PWM:
pwm_hz: 5000
output_pin: NO_PIN
enable_pin: NO_PIN
disable_with_s0: false
s0_with_disable: true
speed_map: 0=0% 10000=100%

MitchBradley, 2022-01-28:
The detailed behavior of helical interpolation with rotary axes differs between controllers; in fact helical interpolation itself is often optional. We will add this feature to the wish list, but it is low on our priority list since it is rarely used, according to this excerpt from the MachMotion manual (the LinuxCNC manual includes a similar statement).

> If a line of code makes an arc and includes rotational axis motion, the rotational axes turn at a constant rate so that the rotational motion starts and finishes when the XYZ motion starts and finishes. **Lines of this sort are hardly ever programmed**

Many CAM programs are able to break motions like this into small linear segments.  Perhaps that would be a workaround that you could use.

ASHAPURAAHM, 2022-01-28:
Thanks for your reply

We are able to do same with 

grbl mega 5x

But it does not have sd card and Marco support

commit 51ddf401 (2022-02-06): Merge pull request #263 from bdring/Helical

Move all axes in helical interpolation, fixes #262
commit 84362538 (2022-02-06): Move all axes in helical interpolation, fixes #262
closed 2022-02-06 by MitchBradley