bdring/FluidNC #264: Job killed while opening Webui in another device.

prak1111, 2022-01-29:
Hello,
A job was running on machine and the device I used to start the job, got an issue, so I connect the machine using another device(to check the status). So just after connecting, the machine quit the current running job. It results in restarting the job.
Is it a bug? Or there an option that I need to enable?

jelsdon, 2022-01-30:
Hi @prak1111  - Did the job restart or did it mark as finish mid way? Is it running from SD Card?
I've currently got an issue where my job running from SD will be marked as complete without reaching the end of a file. Last time it occurred I'd just reopened the WebUI.

jelsdon, 2022-01-30:
Just confirming that I can replicate, have had it from versions 3.2.9 through to current latest 3.3.1
1. Run job from SD via WebUI
2. Close WebUI
3. Open page to WebUI
4. Job is marked as completed and machine marked as idle

Of concern is the spindle left running / laser firing

prak1111, 2022-01-30:
Hello All,
The job was running from sd card. The job was showing succeeded in mid way. And machine became idle. Coordinates shown in webUi was correct.
Spindle running manually so don't know about the signal status.
I open webUi in my phone. Letting it connected, I open the webUi In my pc. The log shows- job succeeded and machine became idle.

gorlovivan, 2022-02-04:
Have same problem. But not shure if this happens on connecting with other browser (phone in my case). But sam job is stoping everytime in random place, coordinates is correct (so i can go to zeroes for all axes and restatr same job). In command panel is something like this:
```
<Run|MPos:351.159,-56.106,-20.000|FS:320,10000|Pn:PXYZ|SD:32.57,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
<Run|MPos:346.712,-53.725,-20.000|FS:320,10000|Pn:PXYZ|SD:32.58,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
<Run|MPos:347.869,-48.775,-20.000|FS:320,10000|Pn:PXYZ|SD:32.58,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
<Run|MPos:352.578,-48.397,-20.000|FS:320,10000|Pn:PXYZ|SD:32.61,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
<Run|MPos:354.522,-51.825,-18.431|FS:749,10000|Pn:PXYZ|SD:32.69,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
<Idle|MPos:354.522,-51.825,-17.941|FS:0,10000|Pn:PXYZ|WCO:0.000,0.000,0.000|SD:32.69,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
<Idle|MPos:354.522,-51.825,-17.941|FS:0,10000|Pn:PXYZ|SD:32.69,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
<Idle|MPos:354.522,-51.825,-17.941|FS:0,10000|Pn:PXYZ|SD:32.69,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
<Idle|MPos:354.522,-51.825,-17.941|FS:0,10000|Pn:PXYZ|SD:32.69,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
<Idle|MPos:354.522,-51.825,-17.941|FS:0,10000|Pn:PXYZ|Ov:160,100,100|A:S|SD:32.69,/sd/projects/CNC7010/axisPlates/20211107_YAxisPlate_16d5mm_t2.nc>
```
Note that i can't go zeroes any axis before restart firmware - it's idle but nothing happens when i try "→X0, →Y0, →Z0", just getting http 200 OK from web server.

When this happens i restart firmware, goes to zero all axis, restart job.. Tried same job 4 times, never completed =)

If i could help to find what is going on please contact me on https://t.me/gorlovivan

gorlovivan, 2022-02-04:
> 

maybe there's some way i can see detailed log?.. I see FluidNC Settings / Message Level, set it to Debug. But can't see anything else in command panel or in telnet terminal.

tschundler, 2022-02-06:
Similar here:

```
$G
<Run|MPos:-375.825,77.895,-12.167|FS:512,0|SD:inf,/sd/windows-row1.8.ngc>
<Run|MPos:-375.859,58.152,-12.210|FS:512,0|SD:inf,/sd/windows-row1.8.ngc>
<Run|MPos:-375.909,32.485,-12.260|FS:512,0|SD:inf,/sd/windows-row1.8.ngc>
<Run|MPos:-361.042,26.916,-12.310|FS:384,0|SD:inf,/sd/windows-row1.8.ngc>
<Run|MPos:-341.771,26.899,-12.358|FS:384,0|SD:inf,/sd/windows-row1.8.ngc>
<Run|MPos:-322.567,26.865,-12.408|FS:384,0|SD:inf,/sd/windows-row1.8.ngc>
<Run|MPos:-305.540,28.958,-12.455|FS:384,0|SD:nan,/sd/windows-row1.8.ngc>
<Run|MPos:-305.523,48.364,-12.493|FS:384,0|SD:nan,/sd/windows-row1.8.ngc>
<Run|MPos:-305.506,67.416,-12.530|FS:384,0|Ov:100,100,100|SD:nan,/sd/windows-row1.8.ngc>
<Run|MPos:-305.489,86.670,-12.570|FS:384,0|WCO:-379.285,23.585,-10.153|SD:nan,/sd/windows-row1.8.ngc>
<Run|MPos:-305.456,105.925,-12.608|FS:384,0|SD:nan,/sd/windows-row1.8.ngc>
[MSG:/sd/windows-row1.8.ngc file job succeeded]
ok
<Run|MPos:-316.779,114.396,-12.653|FS:422,0|Ov:110,100,100>
<Run|MPos:-338.126,114.396,-12.708|FS:422,0>
<Run|MPos:-359.355,114.396,-12.760|FS:422,0>
[...removed finishing last buffered commands...]
<Idle|MPos:-307.228,28.620,-11.035|FS:0,0>
```

At first I thought the device was panicking and rebooting. But watching the output, it clearly suddenly decided the job is done, wraps up what it was doing and goes idle. All coordinates are correct, and so if I remove some of the completed gcode and start again, everything is good.

Even closing the tab and reopening can do it.

Curiously: when reconnecting, it lost track of the percent complete - you see it became inf/nan. Maybe that's related?

commit 868f4625 (2022-02-06): Run aborted (#278)

* Fixed WebUI hover problem

* Fix #264 - Job killed while opening WebUI
closed 2022-02-06 by bdring