bdring/FluidNC #265: Documentation for NowForever VDF using RS485

LukasGossmann, 2022-01-29:
Here is the documentation for setting up the NowForever VFDs for which support was added in [PR-258](https://github.com/bdring/FluidNC/pull/258)  to work with FluidNC.  
If i missed anything let me know and i'll add it as soon as i can.

### NowForever (RS485)

Manufacturer: Shenzhen NowForever Electronics Technology CO.,LTD.  
Website: http://www.nowforever.cn/  

[image]

The D series of VFDs can be found in the control box of chinese CNC6040 cnc routers.  
Unfortunately there is no manual for the D series so all of this information is based on the manual for the E series.  
However there is a comparison chart made by someone else that shows that almost all parameters are the same.

The manual for the E series can be found online in various places by searching for *nowforever e100 manual* using your preferred search engine.  
Comparison chart (german, also contains link for e series manual): http://moh-computer.de/frequenzumformer-parameter/

The current implementation for supporting NowForever VFDs has been tested against the D100S1R5B (D series, 1PH AC 220V 50/60hz input, 8A output) inverter.  
It should work for other VFDs in the D series too as well as the E series since its manual has been used for reference of the parameters and protocol details.

If controlling the VFD through RS485 isn't working for some reason both the D and E series support a 0-10V interface as well.  
This includes controlling the direction trough another input of the VFD. (See manual for details)

Here is a selection of the parameters needed to make the VFD (D and E series) communicate with FluidNC:

**Selection of RS485 as control and frequency source:**

| Register | Value | Description                     | Possible Values                                                                                                                         |
| -------- | ----- | ------------------------------- | --------------------------------------------------------------------------------------------------------------------------------------- |
| P0-000   | 2     | Command source                  | 0: Keypad<br>1: Control inputs<br>2: RS485                                                                                              |
| P0-001   | 0     | Frequency source                | 0: main frequency source<br>1: auxiliary frequency source<br>2: main + aux<br>3: max(main, aux)<br>4: selectd by control input          |
| P0-002   | 6     | Main frequency source selection | 0: Keypad Potentiometer<br>1: Keypad Up Down Arrow<br>2: AIN1<br>3: AIN2<br>4: Multistep speed<br>5: PID<br>6: RS485<br>7: Internal PLC |

**RS485 parameters:**

| Register | Value                             | Description    | Possible Values                                                                                                                                                  |
| -------- | --------------------------------- | -------------- | ---------------------------------------------------------------------------------------------------------------------------------------------------------------- |
| P0-055   | any free address between 1 and 31 | Address of VFD | 1- 31: slave addresses<br>2: master address                                                                                                                      |
| P0-056   | 2 works just fine                 | Baudrate       | 0: 2400bps<br>1: 4800bps<br>2: 9600bps<br>3: 19200bps<br>4: 38400bps                                                                                             |
| P0-057   | 0                                 | Data framing   | 0:1 start bit, 8 data bits, no parity, 1 stop bit<br>1: 1 start bit, 8 data bits, even parity, 1 stop bit<br>2: 1 start bit, 8 data bits, odd parity, 1 stop bit |

**Setting min and max speed:**

| Register | Value                            | Description   | Possible Values       |
| -------- | -------------------------------- | ------------- | --------------------- |
| P0-007   | Whatever your spindle can handle | Max frequency | Min frequency - 600hz |
| P0-008   | Whatever your spindle can handle | Min frequency | 0 - Max frequency     |

**The following registers are read / written to by FluidNC:**

**Read access only:**

| Register | Description                                                                                                                                                                                                         |
| -------- | ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- |
| 0x007    | Max frequency in hz * 100, same as config parameter P0-007                                                                                                                                                          |
| 0x008    | Min frequency in hz * 100, same as config parameter P0-008                                                                                                                                                          |
| 0x300    | Current fault number<br>0 = no fault<br>1-18 = fault number                                                                                                                                                         |
| 0x500    | VFD status<br>Bit 0: run, 1=run, 0=stop<br>Bit 1: direction, 1=ccw, 0=cw<br>Bit 2: control, 1=local, 0=remote<br>Bit 3: sight fault, 1=fault, 0=no fault<br>Bit 4: fault, 1=fault, 0=no fault<br>Bit 5-15: reserved |
| 0x502    | Current output frequency in hz * 100                                                                                                                                                                                |

**Write access only:**

| Register | Description                                                                                                                                                         |
| -------- | ------------------------------------------------------------------------------------------------------------------------------------------------------------------- |
| 0x900    | VFD control<br>Bit 0: run, 1=run, 0=stop<br>Bit 1: direction, 1=ccw, 0=cw<br>Bit 2: jog, 1=jog, 0=stop<br>Bit 3: reset, 1=reset, 0=dont reset<br>Bit 4-15: reserved |
| 0x901    | Speed to be set in hz * 100                                                                                                                                         |

Example YAML config:

```yaml
NowForever:
  uart:
    txd_pin: gpio.17
    rxd_pin: gpio.4
    rts_pin: gpio.16
    baud: 9600
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 24000=100%
```

bdring, 2022-01-29:
Added

closed 2022-01-29 by bdring

MitchBradley, 2022-01-29:
Nicely written