bdring/FluidNC #266: Machine moving beyond the limits

prak1111, 2022-01-30:
Hi,
After homing the machine I was trying to move the gantry. By mistake I clicked the negative direction (-100mm) move and machine started to move in negative direction, breaking the limit switch and pulling the flexible coupling like spring.
How can I make sure machine won't move beyond the specified length in + and - directions.

krishchan007, 2022-01-30:
Did you enable soft limits in config file?

prak1111, 2022-01-30:
Nope. But I am using pulloff 3mm.
So enabling soft limits will do the resolution?

krishchan007, 2022-01-30:
I don't understand what you meant by resolution but, enabling soft limits will top the machine from moving outside the bound limits defined by config file.
And pulloff has nothing to do with restriction of movement, at least to my knowledge

MitchBradley, 2022-01-30:
This is documented in the wiki https://github.com/bdring/FluidNC/wiki/FluidNC-Homing#soft-and-hard-limits

closed 2022-01-30 by MitchBradley

prak1111, 2022-01-30:
I am setting the soft limits to true, but checking it in webUI shows it false.

My config.yaml looks like-

```name: "PINCWL"
board: "ESP32 Controller V4"

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 2
  pulse_us: 2
  disable_delay_us: 1

axes:
  shared_stepper_disable_pin: gpio.13:low
  
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 1800
    acceleration_mm_per_sec2: 50
    max_travel_mm: 1840
    
    homing:
      cycle: 2
      mpos_mm: 0
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      soft_limits: true
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.17:high:pd
      pulloff_mm: 3.0
      standard_stepper:
        step_pin: gpio.14
        direction_pin: gpio.2

  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 1300
    acceleration_mm_per_sec2: 50
    max_travel_mm: 910

    homing:
      cycle: 2
      mpos_mm: 0
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      soft_limits: true
      positive_direction: false
    
    motor0:
      limit_neg_pin: gpio.4:high:pd
      pulloff_mm: 3.00
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.15:low

  z:
    steps_per_mm: 50
    max_rate_mm_per_min: 300
    acceleration_mm_per_sec2: 50
    max_travel_mm: 80

    homing:
      cycle: 1
      mpos_mm: 0
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      soft_limits: true
      positive_direction: true
    
    motor0:
      limit_pos_pin: gpio.16:high:pd
      pulloff_mm: 3.0
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33:low
              
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

probe:
  pin: NO_PIN

10V:
  output_pin: gpio.12
  spinup_ms: 5000
  spindown_ms: 5000
  tool_num: 0
  speed_map: 0=0% 10000=40% 18000=75%

control:
  feed_hold_pin: gpio.21:low
  cycle_start_pin: gpio.22:low

start:
  must_home: true```

Don't know what is wrong?

prak1111, 2022-01-30:
Got it now. was defining the element under wrong section.
Thanks a lot.