bdring/FluidNC #268: Individual step resolution for dual motor axis

mikey526, 2022-02-02:
Hi,
My CNC is a gantry design with two motors on the Y axis. I love FluidNC's auto-squaring feature but there is one further improvement that would really help.
The two Y-axis lead screws have slightly different pitches, and after 1000mm travel there is a 0.8mm difference between the two sides of the gantry. I have done lots of tests to prove that the problem is definitely the lead screws.
Can you add support to set the **steps_per_mm** parameter by motor instead of by axis? This would allow users to correct for poorly matched lead screws and would be a great new feature for gantry type machines.

bdring, 2022-02-02:
That is something we are unlikely to add. It would be a lot of work and spaghettify the code. It would be cheaper to buy better lead screws.

It might be possible via kinematics, but still a lot of work.

mikey526, 2022-02-02:
Buying better ones is not really an option. Nobody sells matched lead screws and the standard ones are no better than what I already have. The high precision ones are too expensive for a hobby level machine and would require modding to fit the new blocks. Software is definitely the way to go.

MitchBradley, 2022-02-02:
The way it currently works is that the step generation works at the axis level and the steps for the axis are applied to all the motors for that axis.  We might change that eventually but it is not on the immediate roadmap.  So don't hold your breath.

closed 2022-02-02 by MitchBradley