bdring/FluidNC #285: WEB UI

sergerzet, 2022-02-09:
The results of engine operation are not yet displayed on the "CONTROLS" and "GRBL" tabs: coordinates, operating modes. V3.3.2

sergerzet, 2022-02-10:
When compiled in VS Code, the "Controls" and "GRBL" tabs work until the "Config.yaml" on the other.

sergerzet, 2022-02-10:
When downloading from Fluid C, the file "config.yaml. Other configuration files don't work. Files with stepper motor control by I2S are considered.

MitchBradley, 2022-02-10:
I do not understand any of this.  Perhaps you can write it in your native language and a human can translate it to understandable English.

sergerzet, 2022-02-10:
На вкладке "Controls" не отображаются показания перемещения двигателей, на вкладке "GRBL" режим "IDLE" не меняется на режим "RUN". Это происходит при замене файла "Config.yaml" на файл "6_Pack_stepstick_XYZABC.yaml". Остальные элементы интерфейса работают нормально. Файл "6_Pack_stepstick_XYZABC.yaml":
board: 6 Pack
name: 6 Pack StepStick XYZABC
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        ms3_pin: i2so.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        ms3_pin: i2so.6
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm:1.000
      stepstick:
        ms3_pin: i2so.11
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.14
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  b:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.19
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
        
  c:      
    steps_per_mm: 808.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.25
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.22
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: i2so.24
  mist_pin: i2so.27
  delay_ms: 0

probe:
  pin: gpio.39
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.14
  enable_pin: gpio.13
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%

MitchBradley, 2022-02-11:
I am working on a possible solution.  I will have more to report tomorrow.

sergerzet, 2022-02-11:
Ok

sergerzet, 2022-02-13:
Messages after a reset:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.0.x (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:6_Pack_stepstick_XYZABC.yaml]
[MSG:INFO: Machine 6 Pack StepStick XYZABC]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 6]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:     Neg Limit gpio.33]
[MSG:INFO: Axis Y (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:     Neg Limit gpio.32]
[MSG:INFO: Axis Z (-150.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:     Neg Limit gpio.34]
[MSG:INFO: Axis B (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO:     Neg Limit gpio.2]
[MSG:INFO: Axis C (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]
[MSG:INFO:     Neg Limit gpio.25]
[MSG:INFO: FeedHold gpio.36]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.13 Out:gpio.14 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Flood coolant I2SO.24]
[MSG:INFO: Mist coolant I2SO.27]
[MSG:INFO: Probe Pin: gpio.39]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.11 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.0 [FluidNC v3.0.x (noGit) (wifi) '$' for help]

sergerzet, 2022-02-17:
I checked version 3.4.0. Everything works fine. The questions are removed.

PR #290 (2022-02-11): Tablet UI handles 6 axes without jamming up
Probably fixes #285
closed 2022-02-17 by sergerzet