bdring/FluidNC #287: I2SO bug when using as Spindle Enable signal

robercy, 2022-02-09:
Hello !
When using I2So outputs to enable Spindles, other I2SO outputs flash randomly, it causes too many undue step pulses, and erratic axes movements. I use TMC2209 as drivers, and UART to control.

To solve this problem I commented the line below (line 70), in the PWMSpindle.cpp file, and I stopped having this problem.

    void IRAM_ATTR PWM::setSpeedfromISR(uint32_t dev_speed) {
        // set_enable(gc_state.modal.spindle != SpindleState::Disable);
        set_output(dev_speed);
    }

I’m using firmware 3.3.2, and I could repeat the problem changing enable pins from I2SO to gpio pins, on Spindle configuration. Maybe, to solve the problem, calling set_enable just when signal change, could resolve.

 It seems that a refresh rate, when used as an output for the Spindle, is too high, not giving the 74HC595 time to operate properly.

This is my configuration:

board: Roby CNC
name: Roby CNC Multibot V2

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false

stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  
  x:
    steps_per_mm: 400.0
    max_rate_mm_per_min: 1200.0
    acceleration_mm_per_sec2: 100.0
    max_travel_mm: 350.0
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 1
      feed_mm_per_min: 100.0
      seek_mm_per_min: 200.0
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: gpio.39
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2209:
        uart:
          rxd_pin: gpio.18
          txd_pin: gpio.19
          baud: 115200
          mode: 8N1
        addr: 0
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.100
        microsteps: 16
        stallguard: 30
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        disable_pin: I2SO.0
        direction_pin: I2SO.1
        step_pin: I2SO.2
        
  y:
    steps_per_mm: 400.0
    max_rate_mm_per_min: 1200.0
    acceleration_mm_per_sec2: 100.0
    max_travel_mm: 450.0
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 1
      feed_mm_per_min: 100.0
      seek_mm_per_min: 200.0
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: gpio.35
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2209:
        addr: 1
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.100
        microsteps: 16
        stallguard: 30
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        disable_pin: I2SO.4
        direction_pin: I2SO.5
        step_pin: I2SO.6
    motor1:
      limit_neg_pin: gpio.32
      limit_pos_pin: gpio.33
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2209:
        addr: 2
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.100
        microsteps: 16
        stallguard: 30
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        disable_pin: I2SO.8
        direction_pin: I2SO.9
        step_pin: I2SO.10

  z:
    steps_per_mm: 400.0
    max_rate_mm_per_min: 1000.0
    acceleration_mm_per_sec2: 100.0
    max_travel_mm: 60.0
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 1
      feed_mm_per_min: 100.0
      seek_mm_per_min: 200.0
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.25
      limit_pos_pin: gpio.26
      hard_limits: true
      pulloff_mm: 1.000
      tmc_2209:
        addr: 3
        r_sense_ohms: 0.110
        run_amps: 1.200
        hold_amps: 0.100
        microsteps: 16
        stallguard: 30
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        disable_pin: I2SO.12
        direction_pin: I2SO.13
        step_pin: I2SO.14

PWM:
  pwm_hz: 5000
  output_pin: gpio.23
  enable_pin: I2SO.16
  direction_pin: I2SO.17
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0% 9000=100%
  
Laser:
  pwm_hz: 5000
  output_pin: gpio.15
  enable_pin: I2SO.18
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 10
  speed_map: 0=0.000% 255=100.000%  

Relay:
  output_pin: I2SO.19
  
coolant:
  flood_pin: I2SO.20
  mist_pin:  I2SO.21
  delay_ms: 0
  
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
        
probe:
  pin: gpio.27:low:pu
  check_mode_start: true
  
  
Thanks !

MitchBradley, 2022-10-08:
Try I2S_STATIC mode and see if the problem still occurs in that mode.

Eric-ERGS, 2022-10-08:
Mitch,

I sould have mentioned that I tried that before noticing this bug report
and going to static had no effect the outputs still flickered.

Thanks,
Eric.

robercy, 2022-10-10:
Hello, Eric and Mitch.
Nice to hear , that this problem have a solution. Tomorow I will  try it.

Keep the good work, fellows.

    Greetings.

robercy, 2022-10-10:
Hi Mitch,

Version 3.6.3 tested on my setup, with I2S_STREAM, and it's working now. Congratulations!

Have a nice day, and Thanks.

closed 2022-10-10 by MitchBradley