bdring/FluidNC #301: Dual functions for reset and cycle_start pins

cotepat, 2022-02-16:
A lot of simple pendants or button boxes are have "Stop" and "Start" buttons. I always hesitate on how to use those... I usually end up using the stop button as a reset button and the start button as a cycle start button... so I have to rely a lot on the UI for some basic functions (namely unlock and feedhold).

The problem is : when you hit reset, you have to go to the UI to unlock (there is no unlock_pin in fluidnc, and unlock "$X" is not configurable as a macro because macros cannot run when machine is locked). And, of course, you have to restart the job completely...

Buttons and pins are often limited on controllers / 

The idea here is to have start / stop buttons that combine reset / feedhold and unlock / cycle_start functions.

I have it working perfectly with very simple changes in the code. See associated PR...

Here is how it works:

- If "control/dual_mode_reset_pin" is true in config file, then reset_pin will act as feedhold when active state is "Cycle" or "Jog". Otherwise the original reset behaviour will be used). To completely kill a running cycle or jog, just press reset twice and state will change from Cycle/Jog->Hold->Idle. I think this is a good option because there will be a "soft stop" (feedhold) whenever possible, and a "hard stop" (reset) when required (for instance when homing or probing).

- If "control/dual_mode_cycle_start_pin" is true in config file, then cycle_start_pin will act as an unlock pin when state is Alarm. Otherwise the original behaviour of cycle_start_pin will be used). This is very useful: no need to go to the UI to unlock and Alarm, and it does so without using a precious input pin / button.

cotepat, 2022-02-18:
I found a bug in a very specific situation, I will come back with a new pull request when it's fixed.

Fcsteve777, 2022-07-04:
I like this idea, saves on IO pins

closed 2022-08-17 by bdring