bdring/FluidNC #309: ERR: Setting up pin [gpio.19] failed. Details: Pin is already used

jaysettle, 2022-02-21:
Guys thanks for the progressive work on FluidNC. I'm moving from esp32 grbl and have my automated config.yaml (made from your automated service) uploaded however I'm getting some errors.

```
**** Connected to COM3 @ 115200 baud ****
[MSG:INFO: FluidNC v3.4.0]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
An unexpected error was detected: [MSG:ERR: Configuration error at /machine/axes/x/motor1/stepstick: ERR: Setting up pin [gpio.19] failed. Details: Pin is already used.]
An unexpected error was detected: [MSG:ERR: Configuration error at /machine/axes/y/motor0/stepstick: ERR: Setting up pin [gpio.19] failed. Details: Pin is already used.]
An unexpected error was detected: [MSG:ERR: Configuration error at /machine/axes/z/motor0/stepstick: ERR: Setting up pin [gpio.19] failed. Details: Pin is already used.]
An unexpected error was detected: [MSG:ERR: Configuration error at /machine/spi: ERR: Setting up pin [gpio.19] failed. Details: Pin is already used.]
[MSG:WARN: gpio.34 does not support :pu attribute]
[MSG:WARN: Ignored key spinup_ms]
[MSG:WARN: Ignored key spindown_ms]
[MSG:WARN: Ignored key speeds]
[MSG:WARN: Ignored key direction_pin]
[MSG:INFO: Using StepStick Mode]
An unexpected error was detected: [MSG:ERR: Validation error at /spi: SPI MISO pin should be configured once]
[MSG:INFO: Configuration is invalid]
[MSG:INFO: Using default configuration]
abort() was called at PC 0x401acd2b on core 1
ELF file SHA256: 0000000000000000
Backtrace: 0x4008a244:0x3ffb1bf0 0x4008a4c1:0x3ffb1c10 0x401acd2b:0x3ffb1c30 0x401ad708:0x3ffb1c50 0x401acc16:0x3ffb1c70 0x4010d006:0x3ffb1ce0 0x4010d1c9:0x3ffb1d20 0x401ac34f:0x3ffb1dd0 0x400e435f:0x3ffb1df0 0x400d5db8:0x3ffb1e10 0x400d5a43:0x3ffb1e30 0x400d5a79:0x3ffb1e50 0x400db64a:0x3ffb1e70 0x400db67d:0x3ffb1e90 0x400dc405:0x3ffb1eb0 0x400dcbb1:0x3ffb1f30 0x400dd663:0x3ffb1f60 0x4010c82a:0x3ffb1fb0 0x4008c172:0x3ffb1fd0
Rebooting...
[MSG:INFO: FluidNC v3.4.0]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:atflag]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.3.59]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.4 [FluidNC v3.4.0 (wifi) '$' for help]
>>> $$
$10 = 1    (Status report options, mask)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
```

I'm also using the precompiled method for simplicity.

I might be missing a step. I have so far:
1. Flashed a new ESP-WROOM-32D with the precompiled firmware.
2. Uploaded my automated machine file using the webGUI

What am I missing thanks.

jaysettle, 2022-02-21:
My config file...
```
name: XXYZ_4XInput_4XInput_Relay_Spindle_5VOutput_ExtSteppers2
board: 6-pack

kinematics:
  Cartesian:

stepping:
  engine: I2S_stream
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0

axes:
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_all_pin: gpio.33:low
      hard_limits: false
      stepstick:
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0
        reset_pin: gpio.19
    motor1:
      stepstick:
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7
        reset_pin: gpio.19
  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_all_pin: gpio.32:low
      hard_limits: false
      stepstick:
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8
        reset_pin: gpio.19
    motor1:
      null_motor:
  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: false
      stepstick:
        direction_pin: i2so.12
        step_pin: i2so.13
        disable_pin: i2so.15
        reset_pin: gpio.19
    motor1:
      null_motor:
  a:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      null_motor:
    motor1:
      null_motor:
  b:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      null_motor:
    motor1:
      null_motor:
  c:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      null_motor:
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:

coolant:
  flood_pin: i2so.27
  mist_pin: i2so.26
  delay_ms: 1000.000

probe:
  pin: gpio.34:pu
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speeds: 0=0.0% 1000=100.0%
  direction_pin: gpio.0
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 0
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false 
```

jaysettle, 2022-02-21:
Sorry, and here's my original machine file from esp32 grbl.
```
#pragma once

#define MACHINE_NAME            "XXYZ_4XInput_4XInput_Relay_Spindle_5VOutput_ExtSteppers2"

#define N_AXIS 6

// I2S (steppers & other output-only pins)
#define USE_I2S_OUT
#define USE_I2S_STEPS
//#define DEFAULT_STEPPER ST_I2S_STATIC

#define I2S_OUT_BCK      GPIO_NUM_22
#define I2S_OUT_WS       GPIO_NUM_17
#define I2S_OUT_DATA     GPIO_NUM_21

#define X_DISABLE_PIN           I2SO(0)
#define X_DIRECTION_PIN         I2SO(1)
#define X_STEP_PIN              I2SO(2)

#define X2_DIRECTION_PIN        I2SO(4)
#define X2_STEP_PIN             I2SO(5)
#define X2_DISABLE_PIN          I2SO(7)

#define Y_DISABLE_PIN           I2SO(8)
#define Y_DIRECTION_PIN         I2SO(9)
#define Y_STEP_PIN              I2SO(10)

#define Z_DIRECTION_PIN         I2SO(12)
#define Z_STEP_PIN              I2SO(13)
#define Z_DISABLE_PIN           I2SO(15)

#define STEPPER_RESET           GPIO_NUM_19

// Socket #1
#define X_LIMIT_PIN             GPIO_NUM_33
#define Y_LIMIT_PIN             GPIO_NUM_32
#define Z_LIMIT_PIN             GPIO_NUM_35
#define PROBE_PIN               GPIO_NUM_34

// Socket #5
#define SPINDLE_TYPE            SpindleType::LASER
#define SPINDLE_OUTPUT_PIN      I2SO(24) // No PWM
#define SPINDLE_DIR_PIN         I2SO(25)
#define COOLANT_MIST_PIN        I2SO(26)
#define COOLANT_FLOOD_PIN       I2SO(27)

// === Default settings
#define DEFAULT_STEP_PULSE_MICROSECONDS I2S_OUT_USEC_PER_PULSE
#define DEFAULT_HOMING_SQUARED_AXES bit(X_AXIS)

```

MitchBradley, 2022-02-21:
Try removing the reset_pin: definition from the x motor1, y motor0 and z motor0 sections.

MitchBradley, 2022-02-21:
FluidNC lets you define a per-motor reset pin but not a shared one, whereas Grbl_Esp32 only had a shared one.  That is the cause of the confusion.  We need to think about what to do about this.  So far you are the first person to notice it, which just shows how rarely the stepper reset feature is used.

jaysettle, 2022-02-21:
Thanks for that. I have done that. 

I'm getting something different now.. "An umexpected command" at the bottom.

```**** Connected to COM3 @ 115200 baud ****
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.4 [FluidNC v3.4.0 (wifi) '$' for help]
>>> $$
Grbl 3.4 [FluidNC v3.4.0 (wifi) '$' for help]
>>> $G
$10 = 0    (Status report options, mask)
ok
>>> $$
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> $G
$10 = 0    (Status report options, mask)
ok
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
```

bdring, 2022-02-21:
What terminal are you using. Is that part of a gcode sender?

I don't think that error text comes from FluidNC. It might be that the sender got an extra "ok" that it was not expecting. Did you also have the WebUI open?

jaysettle, 2022-02-21:
That is the console of UGS v2.0. I had to leave for work but in the time I had I was unsuccessful at jogging the machine around. Makes sense what you're saying though. I'll try again when I get home.

jaysettle, 2022-02-22:
Ok thanks guys, I'm still seeing some odd stuff. I got another GPIO 19 error so I deleted reset_pin: gpio.19 in motor0 as well as the others.

[image]

```
**** Connected to COM3 @ 115200 baud ****
[MSG:INFO: FluidNC v3.4.0]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: gpio.34 does not support :pu attribute]
[MSG:WARN: Ignored key spinup_ms]
[MSG:WARN: Ignored key spindown_ms]
[MSG:WARN: Ignored key speeds]
[MSG:WARN: Ignored key direction_pin]
[MSG:INFO: Machine XXYZ_4XInput_4XInput_Relay_Spindle_5VOutput_ExtSteppers2]
[MSG:INFO: Board 6-pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 dectect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:250ms]
[MSG:INFO: Axis count 6]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:     All Limit gpio.33:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:     All Limit gpio.32:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:     All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis A (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis B (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis C (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:ERR: Laser PWM frequency is 0.]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant I2SO.27]
[MSG:INFO: Mist coolant I2SO.26]
[MSG:INFO: Probe Pin: gpio.34]
[MSG:INFO: Connecting to STA SSID:atflag]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.3.59]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
An unexpected error was detected: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401b3ae8  PS      : 0x00060130  A0      : 0x800ed802  A1      : 0x3ffb1f00  
A2      : 0x3ffb559c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x0000012c  A7      : 0x00000000  A8      : 0xfffffff0  A9      : 0x00000000  
A10     : 0x3ffbbbe4  A11     : 0x00000000  A12     : 0x3ffbde28  A13     : 0x00000000  
A14     : 0x3ffe4a00  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffffff0  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
ELF file SHA256: 0000000000000000
Backtrace: 0x401b3ae8:0x3ffb1f00 0x400ed7ff:0x3ffb1f20 0x400ed117:0x3ffb1f40 0x400dd99d:0x3ffb1f60 0x4010c83c:0x3ffb1fb0 0x4008c172:0x3ffb1fd0
Rebooting...
[MSG:INFO: FluidNC v3.4.0]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:atflag]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.3.59]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.4 [FluidNC v3.4.0 (wifi) '$' for help]
>>> $$
$10 = 0    (Status report options, mask)
ok
```

Not sure what to make of this yet...

bdring, 2022-02-22:
Not sure why it rebooted, but you have some errors and warnings in your config file. That could be the cause. 

Can you post your current config file here?

jaysettle, 2022-02-22:
Sure thanks.

bdring, 2022-02-26:
The crash was do to the pwm_hz: 0.

I fixed it by enforcing a range for that value. See this branch

https://github.com/bdring/FluidNC/tree/ConstrainConfigValues

jaysettle, 2022-02-27:
Thanks guys that was it. I have movement now. 

I'm trying to get homing configured now and failing there. I've added the following in different places in the config file but I'm getting some errors. 
```
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100s
```

I'm going through the wiki and found an example.
[image]

But I'm getting a msg error
```
**** Connected to COM3 @ 115200 baud ****
[MSG:INFO: FluidNC v3.4.0]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
An unexpected error was detected: [MSG:ERR: Configuration parse error: Expected a float value like 123.456 Line 1 column 12]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Configuration is invalid]
[MSG:INFO: Connecting to STA SSID:atflag]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.3.57]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.4 [FluidNC v3.4.0 (wifi) '$' for help]
>>> $$
[MSG:INFO: Configuration is invalid. Check boot messages for ERR's.]
$10 = 0    (Status report options, mask)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
```

I've the following so far for homing the x axis which is ganged. (I'll have to home the y axis too).

```
name: XXYZ_4XInput_4XInput_Relay_Spindle_5VOutput_ExtSteppers2
board: 6-pack

kinematics:
  Cartesian:

stepping:
  engine: I2S_stream
  idle_ms: 250
  dir_delay_us: 0
  pulse_us: 4
  disable_delay_us: 0
  
axes:
  x:
    steps_per_mm: 33.300
    max_rate_mm_per_min: 15000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 2357
    soft_limits: false
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100s

    motor0:
      limit_all_pin: gpio.33:low
      hard_limits: false
      stepstick:
        direction_pin: i2so.1
        step_pin: i2so.2
        disable_pin: i2so.0
    motor1:
      stepstick:
        direction_pin: i2so.4
        step_pin: i2so.5
        disable_pin: i2so.7
  y:
    steps_per_mm: 319.500
    max_rate_mm_per_min: 2300.00
    acceleration_mm_per_sec2: 140
    max_travel_mm: 1213
    soft_limits: false
    motor0:
      limit_all_pin: gpio.32:low
      hard_limits: false
      stepstick:
        direction_pin: i2so.9
        step_pin: i2so.10
        disable_pin: i2so.8
    motor1:
      null_motor:
  z:
    steps_per_mm: 200.100
    max_rate_mm_per_min: 3400.00
    acceleration_mm_per_sec2: 500
    max_travel_mm: 150.000
    soft_limits: false
    motor0:
      limit_all_pin: gpio.35:low
      hard_limits: false
      stepstick:
        direction_pin: i2so.12
        step_pin: i2so.13
        disable_pin: i2so.15
    motor1:
      null_motor:
  a:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      null_motor:
    motor1:
      null_motor:
  b:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      null_motor:
    motor1:
      null_motor:
  c:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    motor0:
      null_motor:
    motor1:
      null_motor:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5

control:

coolant:
  flood_pin: i2so.27
  mist_pin: i2so.26
  delay_ms: 1000.000

probe:
  pin: gpio.34:pu
  check_mode_start: false

macros:
  startup_line0: 
  startup_line1: 
  macro0: 
  macro1: 
  macro2: 
  macro3: 

start:
  must_home: true
  check_limits: true
  deactivate_parking: false

user_outputs:

laser:
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speeds: 0=0.0% 1000=100.0%
  direction_pin: gpio.0
  disable_with_s0: false
  s0_with_disable: false
  pwm_hz: 200
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false 
```

MitchBradley, 2022-02-27:
```
      seek_scaler: 1.100
      feed_scaler: 1.100s
```
What's with the 's' at the end of that line?

jaysettle, 2022-02-27:
Sorry you are correct. That worked. Now I'm homing my x and y which is what I want. 

However my x is ganged so I need that homing to home each motor. I currently have:
[image]
But it is not homing both steppers on the x axis. 

Thanks guys almost there.

bdring, 2022-02-27:
Read this wiki page.

https://github.com/bdring/FluidNC/wiki/FluidNC-Homing#axis-squaring

It is best to define the switches of the _neg_ or _pos_ pins, depending on the homing direction.

jaysettle, 2022-02-27:
Thanks, I have the ganged motors with one switch so it looks like I need to add another switch. Is that correct?

bdring, 2022-02-27:
If you want it to square, yes.

closed 2022-03-04 by bdring