bdring/FluidNC #315: Lightburn compatibility

frank8500, 2022-02-25:
Lightburn compatibility.
It is not possible to see the configuration values ​​from lightburn.
Could the code be modified to make this compatible?

bdring, 2022-02-25:
We think our configuration system is vastly better than the old Grbl system. Only a fraction of our settings would be available in a "compatible" mode. We do not believe it is very important to configure the machine via senders. We have many users of Lightburn and the authors have given free licenses to our developers. 

We tried to make it compatible for specific things that might be needed to change at run time, like $10. Are there specific things that Lightburn needs to change every time you use it?

MitchBradley, 2022-02-25:
The only way for this to happen would be for Lightburn to change.  Lightburn has a list of classic Grbl configuration values that it uses to display.  Our set of configuration items includes many for which Grbl has no equivalents, so even if we were to include compatibility items for those items that Grbl has, and that Lightburn would know how to use, it would not be possible to create a coherent FluidNC configuration from that subset.

Perhaps you could ask the Lightburn folks to make a version that handles FluidNC.  Our configuration system is self-describing, in that a sender like LightBurn can ask for a complete list of config items, along with their units and data types, and thus automatically generate a user interface panel populated from those items.  That is in contrast to the Grbl system where the the list of items that you get from Grbl contains no information as to what the items mean.  The sender just has to know in advance that, for example, $100 is steps per mm for the X axis and its value is a floating point number.  If a new version of Grbl were to include additional new numbered items, the sender would have no way to know the meaning, so a new sender version would be required in order to handle them.

closed 2022-02-25 by MitchBradley

frank8500, 2022-02-25:
thank you very much for your reply. I don't really need to configure much from lightburn. I prepare the cutting layers, I modify the frequency of the pwm laser and I send it to cut.
if I can give you my opinion about your work, simply extraordinary.

If my opinion is of any use to you, the webiu is fine, but I think that for day-to-day work a tft with the basic functions and sending the gcode from the sender to the board is faster. Perhaps the workflow is to send the complete gcode from the sender to a volatile memory, and the scheduler takes care of sending it to the laser. In this way, the loss of data in the communication with the sender in real time would not be so critical.
I don't know if Ruida controllers do something like that. considering high laser speeds.
I tell you all this from my most humble opinion.

MitchBradley, 2022-02-25:
"Modify the frequency of the pwm laser" - why do you have to change that?  Typically the pwm frequency is set to a good value for the laser electronics, then the laser power is modified with S words in the GCode file.  Changing PWM frequency often is an unusual use case, so unusual that I have never heard anyone else doing it.

"TFT with basic functions" - we are working on that.

"send the complete gcode from the sender to a volatile memory" - The ESP32 does not have enough RAM for that, except for very small GCode files.  That is why we require an SD card in most cases.  WebUI is not a "sender"; it initiates job runs from SD card.

frank8500, 2022-02-25:
The first thing to say is that I use a c02 laser.Modifying the frequency allows to optimize the cutting of the materials. For example, for methacrylate cutting, 5000Hz allows the edge to be polished with the cut.For plywood and wood in general, a lower frequency is needed, allowing the edges not to char and more power is needed for cutting. something as subtle as varying the frequency from 50hz to 40hz allows the cutting power to be reduced by 2% for the plywood I have tested, leaving a less dark cut.
Right now I have a mks dlc32 board installed with grbl esp32 firmware from makerbase (very limited no source code).I have another identical board, to which I have installed fluidnc for testing.To control the laser source I use the ttl output of the laser connected to the source activation pin L.By controlling the frequency and the minimum activation pulse of the same, you have a fairly precise control.However, the ideal would be for the CO2 laser with two control outputs..1. output connected to the input IN of the source (to indicate the power of the laser), that could be a pwm of a high and fixed frequency simulating an analog voltage..2. output connected to the source activation input L... with a modifiable pwm where the laser shot will be synchronized.
I pass you an excel of my creation where the optimal values ​​of the pwm, frequency, dpi and speed are calculated.Above all, it is intended for high-speed raster.The calculation is really interesting since it allows to optimize the precise shots of the laser.If in a raster at 500,300,etc, dpi , there must be 500 shots per inch , modifying the frequency the shots that are indicated depending on the speed and the minimum period of activation of the laser are obtained .
Do you intend to explore the new mks servo 42c servos, based on open source?They can be controlled by uart , integrated pap controller, and the first tests I have done are very positive.I have a couple of older ones on the way if you'd like to try them out.
Regarding the ram memory, I was thinking of an external i2c, spi, etc. memory, whose cost would be affordable, and would speed up communications.
A normal route of work for laser cutting,engraver,etc, would be the following.
.1. Prepare the design..2. Pass design to laser program ( lightburn , lasergrbl , etc ).3. Send the gcode to the laser.
Sorry, in that process I don't see the need for webUI beyond configuring the controller.With a large gcode, it takes a while to load, when you send it to the laser, the controls have a very high delay.
Always with respect for your work, which, as I told you, seems magnificent to me.

I really see a promising future for your project, I sincerely believe that with time it could compete in other leagues, as long as the daily work of the professional who is going to use the machine is taken into account.
Why not compete with "professional" controllers like RUIDA or OTHERS.The CO2 laser field is very promising.

The existing compatibility would continue to be maintained, encompassing a more professional use of the laser.

By the way, sorry for my English, my language is Spanish and I use the translator.

If what I say is interesting to you and you need my help, I would be happy to help within my limited knowledge.

I currently have a modified k40 c02 laser, and plan to build a larger laser, where high raster speeds would be a must.

regardsFran