bdring/FluidNC #317: Alarm Mode blocks character echo and responses.

bdring, 2022-02-25:
When you are in alarm mode, the character are no longer echoed at the serial terminal. One the alarm is cleared, the characters are then echoed along with the responses.

Here is an example session..

```
[MSG:INFO: '$H'|'$X' to unlock]
$H
ok
G0 X1000
[MSG:INFO: Soft limit on X target:1000.000]
ALARM:2
[MSG:INFO: Reset to continue]
```
Here I send G91 and G0X10 here with no echo or response.
I then send ^X to clear the alarm.
```
[Reset]
ok

Grbl 3.3 [FluidNC v3.3.0 (FeedholdAfterSoftLimit-2505156) (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
ok
G91
error:9

ok
G0 x10
error:9
```

Avataar120, 2022-04-12:
Hi,

Same issue here.
This is annoying when installing & testing when not connected to a machine or when limit switch is hit during job ...
Lightburn cannot communicate anymore with GRBL

Is there already a fix date ?

Thanks for your amazing work !!

Avataar120, 2022-04-13:
Hi,

It is done by purpose in the code (see the do/while below)

`static void protocol_do_alarm() {
    switch (rtAlarm) {
        case ExecAlarm::None:
            return;
        // System alarm. Everything has shutdown by something that has gone severely wrong. Report
        case ExecAlarm::HardLimit:
        case ExecAlarm::SoftLimit:
            sys.state = State::Alarm;  // Set system alarm state
            alarm_msg(rtAlarm);
            report_feedback_message(Message::CriticalEvent);
            protocol_disable_steppers();
            rtReset = false;  // Disable any existing reset
            do {
                // Block everything except reset and status reports until user issues reset or power
                // cycles. Hard limits typically occur while unattended or not paying attention. Gives
                // the user and a GUI time to do what is needed before resetting, like killing the
                // incoming stream. The same could be said about soft limits. While the position is not
                // lost, continued streaming could cause a serious crash if by chance it gets executed.
                pollChannels();  // Handle ^X realtime RESET command
            } while (!rtReset);
            break;
        default:
            sys.state = State::Alarm;  // Set system alarm state
            alarm_msg(rtAlarm);
            break;
    }
    rtAlarm = ExecAlarm::None;
}`

I have commented the loop and it's working for my usage.
Firmware is still going to alarm mode when hitting limit and stops movements.
But I don't know all the consequences of my modification ... :)