bdring/FluidNC #328: Can't connect to Fysetc E4 using .bat

ofek256, 2022-03-02:
Hi, 
I'm trying to install FluidNC on my Fysetc E4 which currently has grbl32 running.
I can't connect to it and the connection times out:
[image]
Ignore COM1 and COM9 - I have no idea why my computer recognizes those, it just does. COM13 is the Fysetc E4.
I tried holding the reset button and releasing while "connecting..." shows up, nothing. Clicked it in the middle of the cycle - also nothing. Unplugged and replugged power - also nothing.
Any idea why this might be happening? I'd love some help. 
Thanks!

bdring, 2022-03-02:
Have you asked Fysetc support yet?

ofek256, 2022-03-02:
I got it working through my laptop. only difference there might be between the two is that my PC is running windows 11 and laptop windows 10. Might that be the issue?

bdring, 2022-03-02:
I have heard the Windows 11 driver for the cheap CH340 usb/serial chips might not work with all the pins. You might look for an updated driver.

bdring, 2022-03-04:
I added a note about this to the troubleshooting section of the wiki.

Closing.

closed 2022-03-04 by bdring