bdring/FluidNC #330: Pause / Resume / Current Line in GCode Status

grbd, 2022-03-05:
Hi,
So one of the ways I'm using FluidNC at the moment is for a CNC Machine.
However in some cases I have jobs that can takes hours to complete.
This is also located on a machine in a shared hackspace

One useful feature would be the ability to Pause a Job / make a note of it's position in the G-Code
Then some how resume the job at a later date (this is milling wood instead of 3d printing)

If there was some way to determine which line number in the G-Code was the last one processed, then in theory I could just manually remove the G-Code from the beginning of the file and resume a job that way.
But I don't know of a way of getting at that info Since the Run messages don't contain that information

```
<Run|MPos:44.513,3.813,20.312|FS:929,0|SD:0.02,/sd/3D Roughing 1.gcode>
```

Duet3d has support for this but I admit the capabilities of the ESP32 are limited by comparison

  * https://docs.duet3d.com/User_manual/Reference/Gcodes#m916-resume-print-after-power-failure

Still I think this would be very useful (even if it's just the line number in the status view) for long running CNC jobs

MitchBradley, 2022-03-05:
use_line_numbers: true

in the config file.

MitchBradley, 2022-03-05:
And add line numbers - N words - to your gcode.  Most CAM posts have an option to add them.

MitchBradley, 2022-03-05:
<Run|MPos:248.747,-266.825,-11.278|Ln:1|FS:3600,0|Pn:PX>

The Ln field will appear if use_line_numbers is true and there is an N word on the GCode line.

MitchBradley, 2022-03-05:
Please feel free to use Duet if you like it better.

closed 2022-03-05 by MitchBradley

grbd, 2022-03-05:
Thanks this seems to work I was looking for this but missed it for some reason
```
use_line_numbers: true
```
I've managed to alter my pre-processor and it works spot on.

bdring, 2022-03-06:
Pausing and restarting a job has a lot of issues.

Unlike most 3D printer firmwares, a "pause" in FluidNC happens very fast. It could be partially though a single gcode command. 

The other problem is machine state and "modes". Gcode sets various modes. If you set things like G21 to be in millimeter mode, that stays current for the rest of the job. This includes spindle modes, spindle speeds, motion modes, feedrates, etc.

Restarting a job at a specific point in gcode will not set the proper spindle mode, feedrate etc that was present before the stop.

The job of bringing the machine to a stop and preparing for a later restart should fall on the gcode sender. You would tell it you want to stop. The first thing it would do is stop sending new gcode lines and let the already sent lines finish. It should then store the position and modes at the end of the job.

Upon restart, it would send the machine to the ending point, reset all the old modes like spindle on and then begin sending gcode again.

grbd, 2022-03-06:
That's fine, I understand what you're saying about the initialisation G-Code, but getting the line number is enough for me
I'm happy with manually hacking the G-Code by removing the code from after the Initialisation (setting the mm, making sure the G54 is set correct etc) and where I left off.

As long as I can roughly tell which line number I got as far as, when milling I can just start it off a bit before where I left it.