bdring/FluidNC #336: Filename too long - Web UI wont launch - Loading...

Gerard1983, 2022-03-15:
Please only submit bugs for the **latest** version of the code. You can check the version number in the startup messages.

**Do not add the bug label**. The developers of this repo will do that after confirming the issue.

Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

Select a COM port
0: COM1 (\Device\Serial0)
1: COM3 (\Device\Silabser0)
Choice: 1
Using COM3
Ctrl-] to exit, Ctrl-U to upload, Ctrl-R to reset
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
Resetting MCU
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.4.1]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Cannot open configuration file:config.yaml]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.1 (wifi) '$' for help]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]

Please answer the following questions.

What version of the firmware are you using?
v3.4.1

Is the problem repeatable?
Yes

Under what conditions does the bug occur?
**When I'm opening the browser UI on my tablet. Works on my PC and Laptop.
My old ESP32_Grbl works on the tablet. The FluidNC worked on my tablet a few days ago. Doesn't work any longer for unknown reasons. The browser says "Loading..." continuously.
I now notice all of the "filename too long" messages everywhere upon reset and about 10 seconds after the Grbl prompt appears.
Will also get two "filename too long' msgs echod in my command prompt every time I refresh my WebUI browser.
The browser on the tablet says "Loading..." and sits there forever, rather than opening the ESP3D interface.
I have erased my dev board using your batch file. Reloaded the firmware. There has been no change to the UI behavior or "Filename too long" messages in the command prompt. Also tried deleting all files from LocalFS except index.html.gz. I reset the name of the config file back to config.yaml. By the way, seems an "erase" doesn't completely erase the ESP32. After reinstalling the firmware I noticed my config file was still the same name in the config, and still appeared in the filesystem. I attached a few files documents my 15 hour ordeal over a 3 day period. Should put you to sleep at night.
By the way, how do you clear the error messages via $E?**

**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```C++
  #define EASIER_TO_READ true
```
[Tablet-browserNotWorking.odt](https://github.com/bdring/FluidNC/files/8254451/Tablet-browserNotWorking.odt)
[Tablet-filenametoolong.odt](https://github.com/bdring/FluidNC/files/8254466/Tablet-filenametoolong.odt)
[Tablet-browserNotWorking.odt](https://github.com/bdring/FluidNC/files/8254471/Tablet-browserNotWorking.odt)

MitchBradley, 2022-03-15:
It will take me some time to digest all the information that you provided - and I have a lot of other things I must do today - but in the meantime:

Have you tried connecting to WebUI from a PC running a modern browser?  It is conceivable that we did something that is incompatible with older browsers that could be found on tablets running ancient versions of Android.  Or something like that.

bdring, 2022-03-15:
There could be some corruption or bad files on the local file system of the ESP32. You might try running **install-fs** script from the release package again. Then reload your config file.

Gerard1983, 2022-03-16:
Thanks very much for your prompt replies guys. The ESP32 board connects to my modern PC, laptop, mobile phone and even a $300 HP windows 8 computer no problem.
I think the issue the tablet is having is this recently discovered "Filename too long" message that comes up.
It doesn't seem to affect anything else except the tablet.
Bart, I did an erase.bat, install-fs.bat, and then install-wifi.bat again.
Seems some things still remain, such as the config file name in firmware, as well as error messages.
I'm currently looking into the ESP32 documentation to see if there is a "restore to factory defaults" from espressif that I can run.

MitchBradley, 2022-03-16:
"Filename too long" means that, in a file pathname like "/spiffs/something_that_is_really_long", the "something.." part is longer than 30 characters.  That restriction has always been there, but apparently something about the combination of index.html.gz running on a tablet is resulting in a "something.." that is now malformed.

One thing to try would be to get some older releases and use just "install-fs" to see if earlier index.html.gz files fix it.

Gerard1983, 2022-03-16:
Thanks Bart and Mitch. 
I did the install-fs again, uploaded the config file etc. Still same issue. Nearly bricked my ESP32 by holding down the boot button after the batch file finished and while also resetting via the terminal. Lost some SPI capability which fed an error to my terminal window over and over. I proceeded to install-wifi.bat anyway for a laugh to see what would happen. Noticed I didn't need to press the boot button anymore, it automatically put the board into boot mode shortly after the "connecting...---"
Did the whole thing again and yep, no need to press boot. Hopefully that helps other people with these Duinotech boards.
As for the tablet, somehow managed to do a google play store update, followed by a Chrome update.
Even though Chrome is updated, it pops up with a prompt to update to the latest version. That's just, wrong.
I'm running Android 4.2.2 (Jelly Bean?) which is also up to date.

I believe when my tablet was working last week, I was using an OS  / Chrome update from around 5 years back.
The "factory resetted" tablets OS and Chrome is too old to work. The current 2022 OS /Chrome updates are too new to work.
Oh well it'll teach me for trying to have a clean slate. If it aint broke don't fix it.

In the end the tablet works with my older ESP32_Grbl board. And I have a spare laptop that works with my newer FluidNC board. I'll be running two builds since I have the hardware for 2 builds anyway.
The issue doesn't seem to be with the FluidNC board, files etc. "Filename too long" only ever comes up when connecting to Wifi/opening browser with the tablet, not any other device. So it's an echo from the tablet of sorts.
Going to use some EI to conclude that having a vendetta against a $100 tablet isn't worth anymore of my time.
Will make a $ contribution to the project.
Much appreciate the help.

MitchBradley, 2022-03-17:
Please try this test release https://github.com/bdring/FluidNC/releases/tag/TestFilenameReport .
It displays the filename when it is too long.  That might help us learn the cause of the error.

Gerard1983, 2022-03-23:
Hi Mitch, Thanks for your continued effort. How would I proceed with the source code once downloaded?
Before I try that, some updates from me.
Bought a brand new ESP32 today, ran the fs and wifi install. All worked well on the PC via wifi, no errors.
As soon as I connected to tablet and opened the browser, filename too long again. Since the browser was stuck at the loading screen, I was wondering if it also had something to do with not running javascript or similar, as I've had this issue before.
Javascript is turned on in Chrome, but maybe it was still not running something.
I tried running NCviewer.com over the internet, and this had issues with the simulation, so I figured this version of Chrome on my tablet is a dud! It won't run software through the browser properly.
Downloaded a new browser "Opera".
Sure enough, it worked straight off the bat.
Turns out google is just another microsoft. garbage.
Now I know the issue only lies in Chrome for this specific tablets operating system, I'm moving forward. As far as I'm concerned problem fixed. At least I have ESP32 spares...

MitchBradley, 2022-03-23:
> How would I proceed with the source code once downloaded?

Why would you need the source code?  The test release that I made for you included the binary and installer package.  I suppose that I wasted my time creating it for you.

That release is no longer available because the latest FluidNC release v3.4.2 includes the code that was in the test release.  That code displays the actual filename as part of a "filename too long" message.  That would have given a clue about what the browser was sending.

imechura, 2022-03-27:
I also have the filename is too long message.

New clean install on a generic esp32 dev board

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.4.1]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine ESP32 Dev]
[MSG:INFO: Board ESP32 Dev]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.13:low]
[MSG:INFO: Axis X (-10000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO: Axis Y (-10000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO: Axis Z (-10000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.2 Dir:NO_PIN]
[MSG:INFO: Using spindle Relay]
[MSG:INFO: Probe Pin: gpio.32:low:pu]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.1 (wifi) '$' for help]
[MSG:DBG: Cmd::Reset]

Grbl 3.4 [FluidNC v3.4.1 (wifi) '$' for help]
[MSG:INFO: Filename too long]

MitchBradley, 2022-03-27:
version 3.4.2 will tell you the actual name that is too long.

MitchBradley, 2022-03-27:
One person using 3.4.2 reported that the long filename was "/spiffs/api/spiffs/v1/spiffs/interap-communication".  I do not know where that filename is coming from.

imechura, 2022-03-27:
Its going to be a while before I can upgrade the firmware version on this system. The webUI comes up but here what is what is displayed in console after about 2 minutes.

[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[... 197 lines cut ...]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]
[MSG:INFO: Filename too long]

MitchBradley, 2022-03-27:
I am working on it.

MitchBradley, 2022-04-01:
We found a root cause for this.  On Google Wifi networks, the access points communicate with each other on port 80, by trying to do HTTP GET /api/v1/interap-communication . Apparently they scan multiple devices on the network looking for a response to that URL.  There was a bug in the FluidNC pathname code.  If a pathname comes in that does not have a /spiffs/ or /sd/ prefix, the pathname code assumes /spiffs/ and replaces the initial / with /spiffs/ .  But instead of replacing just the first /, the replacement code was using the Arduino String replace() method, which replaces all occurrences, so /api/v1/interap-communication became /spiffs/api/spiffs/v1/spiffs/interap-communication, which triggered the Filename Too Long check because that string is longer than the maximum length of a SPIFFS filename.

Gerard1983, 2022-04-17:
> > How would I proceed with the source code once downloaded?
> 
> Why would you need the source code? The test release that I made for you included the binary and installer package. It seems that that I wasted my time creating it for you.
> 
> That release is no longer available because the latest FluidNC release v3.4.2 includes the code that was in the test release. That code displays the actual filename as part of a "filename too long" message. That would have given a clue about what the browser was sending.

Hi Mitch. I understand you spent time creating it to echo the filename to see why it was throwing the error. Can appreciate your work. It's what I would have done, however I already wasted plenty of my time on it... and had no more to give. 
I figured someone else would come along and run the code to contribute and help resolve the issue, as they have.
Open source has always been this way, I mean after all it's free. Though sometimes I wonder is it worth the bugs, time and hassle even though it is free. It's why professionals throw $20,000+ at machines that just work out the box. Which is what I just did. I have a CNC lathe with a Siemans 808D being built now. I think it has saved me money by doing so. I'm no longer young.

I did a few open source CNC projects over the years, and honestly none of them ever really worked. The original Grbl is probably the only firmware that did.
Reason for this I believe is that it was developed up until a point to do one job, then updates performed again and again to iron out all bugs, rather than upgrades being performed for every customers little quirks. As you know upgrades can introduce bugs, leaving you in a never ending cycle of upgrades/updates/frustration. As far as I'm aware, new features and upgrades aren't performed on 8-bit Grbl.
Projects that never end lead to burnout as it never really reaches completion.

My preference would be a system that is generic, with specific tailored releases for a lathe, mill, plasma etc.
(look at the machine.config files? yes but not exactly the limit of my thinking)
I'd release them one by one with set features, update them to perfection, and then leave them alone.
For those who wish to have an edge release full of new code for some fandangled feature, they can accept the issues to gain these features.

imechura, 2022-04-22:
Can you tell me which file has erroneous code so that I can made the change to fix it?

MitchBradley, 2022-04-22:
It is already fixed in release v3.4.3 via commits https://github.com/bdring/FluidNC/commit/2d581c531fe4a671844595ac15a07171e1b4d15f and https://github.com/bdring/FluidNC/commit/643c6dd412dd4d27484278be5a278bbc25a8849a