bdring/FluidNC #338: Failing to connect to wifi

shuff57, 2022-03-16:
I believe I have done my best trying to find a solution to my problem by reading through the wiki, checking legacy grbl esp32 wiki, and checking both "issues" sections. However, it  may be strictly a hardware issue. 

I wanted to flash and use a Wemos D1R32 for a project. I believe I bought it before it was listed as not compatible due to the shield not using 100% of the esp32 pins. I figured since I only wanted to build a 3 axis machine, I wouldn't need all the pins so I pressed on. Now that you have my rationale for using non-compatible hardware...

I don't think using a Wemos D1R32 would prevent the Sta Password from saving. I tried using the fluid term and the WebUI. No matter what I do it appears the password will not save to the memory. (Note: I already removed the resistor from the protoneer cnc shield, flashed and was unable to connect to my local network. Next I removed the shield altogether, erased, refreshed, and I am still unable to connect to my local network

I had extra 38 pin esp32 nodemcu lying around so I flashed it and was able to get my SSIS and password to save without issue. 

Any help on this issue would be appreciated. Attached is a readout from the Wemos D1R32

[image]

MitchBradley, 2022-03-16:
How do you know that the problem is that the password will not save to the memory, as opposed to other possible reasons that connection might fail?

shuff57, 2022-03-17:
I don't know for sure that the password will not save. My only point of reference is that a spare esp32 nodemcu was able to connect and save all the settings without issue. I'm open to brainstorming other reasons the connection failed. I just can't think of any atm.

shuff57, 2022-03-17:
Upon further troubleshooting, I think its the board. I uploaded a simple sketch in arduino that would print the connection status in the serial monitor and it continued to fail. However, I had it working on different boards in the past. I'm not sure how to proceed, its beginning to move past my level of experience. Thank you for the reply!

MitchBradley, 2022-03-17:
Buy another board.   Time is worth much more than the cost of an ESP32 module.  At least mine is...

shuff57, 2022-03-17:
That's the plan! Thank you for your help on this matter!

closed 2022-03-17 by shuff57