bdring/FluidNC #345: Failed to execute script py_rth_multiprocessing

Althafzahra, 2022-03-21:
hello, please help, sir, here I have a problem while installing fluidnc, what is the cause and what is the solution? Thank you
[image]

MitchBradley, 2022-03-21:
Which version of Windows ?

Althafzahra, 2022-03-21:
> Which version of Windows ?

I'm using windows 7 64 bit

MitchBradley, 2022-03-21:
Google found this https://github.com/pyinstaller/pyinstaller/issues/4855#issuecomment-676851508

Althafzahra, 2022-03-23:
> Google found this [pyinstaller/pyinstaller#4855 (comment)](https://github.com/pyinstaller/pyinstaller/issues/4855#issuecomment-676851508)

I have updated and running but I found another problem as shown in the picture. what should i do sir?

Althafzahra, 2022-03-23:
> Google found this [pyinstaller/pyinstaller#4855 (comment)](https://github.com/pyinstaller/pyinstaller/issues/4855#issuecomment-676851508)

[image]

MitchBradley, 2022-03-23:
Microsoft stopped supporting Windows 7 two years ago so we do not support it either. 

The "esptool" program that we use to install FluidNC comes from Espressif, the maker of ESP32 chips.  Search the web for how to install and use it.  The "esptool.exe" variant that we use only works on modern systems, but the basic "esptool.py" version that depends on a Python language setup will probably work on your system.  I do not have time to teach you how to do that, but there are many tutorials on the web about using esptool.

If you can install Python version 3 and get esptool.py from the web, you can then run these commands in a CMD window to install FluidNC:

```
cd fluidnc-v3.4.2-win64
python esptool.py --chip esp32 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 common\bootloader_dio_80m.bin 0xe000 common\boot_app0.bin 0x10000 wifi\firmware.bin 0x8000 wifi\partitions.bin
```

The first `cd` command puts you into the fluidnc installer folder.  You might have to add additional path components to get to that folder.

Althafzahra, 2022-03-24:
> 
Thank you sir for your time and answer, I will try.

Althafzahra, 2022-03-25:
> Microsoft stopped supporting Windows 7 two years ago so we do not support it either.
>
> The "esptool" program that we use to install FluidNC comes from Espressif, the maker of ESP32 chips. Search the [...]

it's been going well, so far haven't found any problems, thank you very much sir for your help. I have another question, sir, does Webui Fluidnc not work on Android?
[image]
trying on android smartphone having problems
[image]

MitchBradley, 2022-03-25:
You should configure FluidNC to use an external Wifi setup - STA mode - instead of using AP mode.  AP mode is only good for initial setup.  It has performance problems for real use.

Once you are connected to the external Wifi, you can access FluidNC from your smartphone if it is also connected to the same Wifi.

Refer to this documentation https://github.com/bdring/FluidNC/wiki/FluidNC-Wifi-and-Bluetooth-Setup

Althafzahra, 2022-04-02:
> You should configure FluidNC to use an external Wifi setup - STA mode - instead of using AP mode. AP mode is only good for initial setup. It has performance problems for real use.
>
> Once you are [...]

I've been trying it for a few days and it's working fine.  thank you sir for the help.

closed 2022-04-03 by MitchBradley