bdring/FluidNC #353: MKS DLC32 WITH NC FLUID FIRMWARE

kweilim98, 2022-03-29:
Hi,i am new here and i wanna to ask is it can i have the fluid nc firmware to my mks dlc32 motherboard? because the board is still new and not more future to the cnc user, that why i decided to change my board firmware into fluid nc , is it possible? any guideline for me thx u.

DirtyEngineer, 2022-03-29:
The example config for the DLC32 board can be found [here.](https://github.com/bdring/FluidNC/blob/main/example_configs/MKS_DLC32_21_XYZ.yaml)

Follow the steps outlined in the [wiki](https://github.com/bdring/FluidNC/wiki/FluidNC-Compiling#using-pre-compiled-files).

kweilim98, 2022-03-30:
> The example config for the DLC32 board can be found [here.](https://github.com/bdring/FluidNC/blob/main/example_configs/MKS_DLC32_21_XYZ.yaml)
> 
> Follow the steps outlined in the [wiki](https://github.com/bdring/FluidNC/wiki/FluidNC-Compiling#using-pre-compiled-files).

The program write is for the laser engraving, it is suitable for my CNC router? Since I can't see the spindle function at there

kweilim98, 2022-03-30:
> The example config for the DLC32 board can be found [here.](https://github.com/bdring/FluidNC/blob/main/example_configs/MKS_DLC32_21_XYZ.yaml)
> 
> Follow the steps outlined in the [wiki](https://github.com/bdring/FluidNC/wiki/FluidNC-Compiling#using-pre-compiled-files).

Can u tag me how to compile the file to my motherboard? I m newbie here thx u

DirtyEngineer, 2022-03-30:
> The program write is for the laser engraving, it is suitable for my CNC router? Since I can't see the spindle function at there

You will have to adapt the config file to your specific setup. The wiki has a section with examples for [spindle](https://github.com/bdring/FluidNC/wiki/Fluid-Spindle-Setup) setups.

Read the entire wiki and you will get a handle on how to properly set up a config file and load it.  All the information is there.  A lot of work was put into the documentation so please consider donating to the project.

kweilim98, 2022-04-01:
is it got the program for the touch screen panel for mks ts35?

bdring, 2022-04-01:
No touch screen yet. You can use a smart phone or tablet with the WebUI.

bdring, 2022-04-01:
We make no promises about future features or dates. 

We have started some work on a general display interface with some display types. I don't know anything about how that display works and have not done anything with it. I would not count on using that display for a long time.

DirtyEngineer, 2022-04-01:
> 

You can use the stock CNC MKS firmware if you want the screen to function. https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware

kweilim98, 2022-04-01:
mean i can burn the firmware fluidnc and mks screen together in a board?

kweilim98, 2022-04-01:
then how sir?  can u tell me more detail?

Skorpi08, 2022-04-01:
https://github.com/makerbase-mks/MKS-DLC32

Product introduce video: https://www.youtube.com/watch?v=U_OzlMxwms8&t=6s
Basic usage for laser engraving machine: https://www.youtube.com/watch?v=MKRBYVbxJmw&t=139s

kweilim98, 2022-04-01:
u are not answer my question sir, i mean how to compile the fluidnc and touch screen panel program on that mks dlc32

Skorpi08, 2022-04-01:
The question has already been answered, and not just once.

MitchBradley, 2022-04-01:
You cannot use fluidnc with the mks display.  If you are unhappy that you paid money for the display, complain to the people you paid.  We received no money from the sale of the MKS board, even though they used a modified version of our previous firmware (Grbl_Esp32).  They did not publish their modified version, which is illegal because the license terms require that they do so.

kweilim98, 2022-04-02:
> You cannot use fluidnc with the mks display. If you are unhappy that you paid money for the display, complain to the people you paid. We received no money from the sale of the MKS board, even though [...]

I see that sir, thx u

growler242, 2022-04-02:
> You cannot use fluidnc with the mks display. If you are unhappy that you paid money for the display, complain to the people you paid. We received no money from the sale of the MKS board, even though [...]

That is sad :( I ordered a MKS board, waiting for it to arrive. I will buy a 6 pack next time!

closed 2022-04-02 by bdring