bdring/FluidNC #363: FluidNC on CNC xPRO v5

lagnat, 2022-04-04:
Steps taken:
1.  Loaded the latest release
2. Used the auto convert feature to generate yaml from the config file in the Spark Concepts GitHub repo.
3. File failed to load.  Had to add spi_index and remove the cs_pin in the other motors to put into daisy-chain mode.
4. File loaded but produced an error about not being able to self-test the motors.
5. Tried to follow FAQ for `Motors don't move`.
    `$stepping/idle_ms=255` gave me errors about trying to re-use pins.
    Didn't see anything called "disable pin" but tried flipping `cs_pin` to `high`.  
    kernel panic

Thanks in advance for any advice.

```
[MSG:INFO: FluidNC v3.4.3]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:v5.yaml]
[MSG:INFO: Machine CNC_xPRO_V5_Machine_Template]
[MSG:INFO: Board unknown]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO:     All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO:     All Limit gpio.34:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:4 R:0.050]
[MSG:INFO:     All Limit gpio.39:low]
[MSG:INFO:   Motor1]
[MSG:INFO: X Axis driver test passed]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e1e25  PS      : 0x00060d30  A0      : 0x800e2050  A1      : 0x3ffb1e40  
A2      : 0x3ffb9984  A3      : 0x00000000  A4      : 0x3ffce71c  A5      : 0x00000000  
A6      : 0x00000707  A7      : 0x80000000  A8      : 0x800e1e19  A9      : 0x3ffb1e20  
A10     : 0x3ffb3848  A11     : 0x00000000  A12     : 0x3ffb97f0  A13     : 0x00000707  
A14     : 0x7ff00000  A15     : 0x7ffc1fff  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x40002390  LEND    : 0x4000239f  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400e1e25:0x3ffb1e40 0x400e204d:0x3ffb1e90 0x400e1a5c:0x3ffb1eb0 0x401b3c2b:0x3ffb1ed0 0x400da90f:0x3ffb1ef0 0x400da33d:0x3ffb1f10 0x400da5ad:0x3ffb1f30 0x400dd9f5:0x3ffb1f60 0x4010cf82:0x3ffb1fb0 0x4008c0ba:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.4.3]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.3 (wifi) '$' for help]
```

bdring, 2022-04-04:
I do not have hardware to test that config file. 

I'll try to decode that backtrace later today.

bdring, 2022-04-04:
This is due to you not having [homing sections](https://github.com/bdring/FluidNC/wiki/FluidNC-Motor-Setup#homing) in your config file.

We will be implementing some fixes to prevent this in the future, but it might take a while before it is released. 

If you want a quick fix, add those sections to your file. If you need help, I can do it for you.

MitchBradley, 2022-04-04:
You could also try re-running the converter.  I just changed it to be more aggressive about adding homing: sections.

lagnat, 2022-04-04:
I manually added the homing sections and it's good to go.  I'll post the final config back here when I'm happy with everything in case anyone else is having similar problems.

MitchBradley, 2022-04-04:
I translated that XPro config file months ago as a test of the converter, but since none of us developers have an XPro, we could not test it live.  One would hope that the XPro folks would have sent us one, considering the number of questions that we end up fielding from their customers, both in Grbl_Esp32/FluidNC land and CNCjs land.

lagnat, 2022-04-04:
Assuming you got the homing in there, the only other things I had to change are listed as steps 3 and 5 in my first post.  I'm making other changes specific to my setup so maybe not a good idea to use my yaml but I'll post it anyway when I'm done.

closed 2022-04-06 by lagnat

MitchBradley, 2022-04-06:
I think that I fixed the SPI problem with a recent checkin.

MitchBradley, 2022-04-10:
I just now fixed the converter.  My last attempt at a fix got the spi_index numbering wrong.  The first spi_index should be1, not 0.