bdring/FluidNC #364: Support for a different PWM Spindle (Simple H-Bridge)

spsspssps, 2022-04-06:
I have successfully tested fluidnc on my custom CNC board https://github.com/spsspssps/spscnc My card uses an H bridge to drive the spindle, and requires 2 non simultaneous PWM signals.

Current prototype schematics (with straps):
https://github.com/spsspssps/spscnc/blob/releases/1.0/1.0.2/doc/design.description.md

The current PWMSpindle driver supports DIRECTION and OUTPUT, My driver would require OUTPUT_A, OUTPUT_B. 
When the spindle is set CW, OUTPUT_A shall output PWM, and OUTPUT_B shall output GND.
When the spindle is set CCW, OUTPUT_A shall output GND, and OUTPUT_B shall output PWM.

I started a feature branch locally, and I could deliver a pull request if this feature is approved, but I have two possible ways to implement this:

a) Extend the PWM class and add a reverse pin, depending on the direction, PWM is set on OUTPUT_A or OUTPUT_B, name the pins _output_pin and _output_b_pin (But I see the BESC controller is already inheriting from this class, and I don't want to make a mess.
b) Add a derivative class (from PWM or ONOffSpindle and call it SimpleHBridge, override several methods from the PWMSpindle class, so that these changes do not affect any other existing driver.

I can write the feature myself, but I would like to agree on how to introduce this in harmony with the rest of the project.

MitchBradley, 2022-04-06:
subclassing is probably best

MitchBradley, 2022-04-06:
The other thing I would ask is that, if your PR is accepted, you lurk on Discord and field any questions related to it, on a long-term basis.  I get tired of supporting other people's code.

spsspssps, 2022-04-06:
Thanks ,
I can agree to lurking every now and then, and to add documentation for the driver both on the source code and on the main documents. I will start with subclassing, and I will try not do deviate too much from the original classes.

closed 2022-08-17 by bdring