bdring/FluidNC #367: xPro V5 - x/y/y1/z axis motor error

possu911, 2022-04-09:
Hi,

First thanks to bdring for this awesome project and your support.

Got the machine.h converted with the tool:
https://github.com/bdring/Grbl_Esp32/issues/1152

Upon loading the config file to xPro v5, I get axis error on all motors.
HW and my setup is working OK (with xPro v5 stock FW)

@Spark-Concepts team; There seem to be many of us here using your product. It would be great if you would post yaml files for your configurations…

Here’s the boot up log:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.4.3]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine CNC_xPRO_V5_Machine_Template]
[MSG:INFO: Board 0407_v0.13]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO: Motor0]
[MSG:INFO: tmc_5160 Step:gpio.12 Dir:gpio.14 CS:gpio.17:low Disable:NO_PIN Index:0 R:0.050]
[MSG:INFO: All Limit gpio.35:low]
[MSG:INFO: Motor1]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO: Motor0]
[MSG:INFO: tmc_5160 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:1 R:0.050]
[MSG:INFO: All Limit gpio.34:low]
[MSG:INFO: Motor1]
[MSG:INFO: tmc_5160 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:2 R:0.050]
[MSG:INFO: Axis Z (0.000,300.000)]
[MSG:INFO: Motor0]
[MSG:INFO: tmc_5160 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:3 R:0.050]
[MSG:INFO: All Limit gpio.39:low]
[MSG:INFO: Motor1]
[MSG:ERR: X Axis driver test failed. Check connection]
[MSG:ERR: Y Axis driver test failed. Check connection]
[MSG:ERR: Y2 Axis driver test failed. Check connection]
[MSG:ERR: Z Axis driver test failed. Check connection]
[MSG:INFO: Door gpio.16:low:pu]
[MSG:INFO: Macro 0 gpio.13:low:pu]
[MSG:INFO: Macro 1 gpio.0:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.4 Out:gpio.25 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Mist coolant gpio.21]
[MSG:INFO: Probe Pin: gpio.22:low:pu]
[MSG:INFO: Connecting to STA SSID:xxxxxxxx]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.51.220]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.3 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]

bdring, 2022-04-09:
You should search on our [Discord](https://discord.gg/4XHS85Ef) for xPro-V5 people and maybe DM them.

possu911, 2022-04-10:
Thanks for your help. 
Ended up copying the config from https://github.com/bdring/FluidNC/issues/363 
Added the homing sections.
No errors on bootup. x/y/z are moving.

bdring, 2022-04-10:
@possu911  Can you post you config file here?

I will add it to the example files.

possu911, 2022-04-11:
@bdring 
The 'config_working' file is one from yesterday & right after made the post above.
Noticed this morning when I powered the system up that steppers were not moving, neither from the web UI or CNCjs.
Fired up fluidterm, and upon start or reset it complaints errors on all axis.
Turned power off from the xPro, took usb cable out. Power on to xpro, hear normal nice hum from the steppers.
Plugged usb cable in. Could control the CNC from web ui and CNCjs.
fluidterm looked good.
When I tried to change the steps_per_mm w CNCjs console, got errors about daisy chained steppers and spi_inder. (sorry for vague, unclear description, should have written it down).
Then modified the working yaml on text editor, just changed the steps_per_mm.
Uploaded the new file with fluidterm, ^R for reset and again the same errors on all axises.
Modified the file some more, always same axis errors.
Then the last one, attached config0.23 - I get the error only on x-axis.
Play time today with the machine is over, will continue tomorrow and can take better notes/debug if needed.

Thanks,
possu911

ps.  My machine is Genmitsy 3020 w. xyz steppers. Getting ready for Openbuilds Lead 1010, which is on its way. I'll put the xpro to that, or send it back to Spark-Concept.

config0.23
config_working

bdring, 2022-04-11:
Do you know if a schematic is available for that controller? It is a mystery box to us.

Does the USB connect without main power or does it need main power to operate?

Have you contacted them?

bdring, 2022-04-11:
If the ESP32 is powered (via USB) before the TMC5160s are powered, you will get startup errors.

possu911, 2022-04-11:
Yeah, I noticed earlier that USB powers ESP32. Toggling the main switch does no good: So on power cycle I power it off & unplug USB. Power on & usb in.
Just send mike@spark-concept an email. Included same files as to this conversation. Looking forward for his reply.

bdring, 2022-04-11:
Have you tried the reset button? Power on everything. Establish a USB connection via fluidterm and click that button.

possu911, 2022-04-11:
Tried couple different ways:
-USB connected, ^R on fluidterm resets
-USB connected, fluidterm open, push reset button on the xPro; see same reset txt on fluidterm as w ^R
On both cases axes error msg
-Plugged power cord to xPro, switched it On, fluidterm still connected
Prints nothing on fluidterm (no surprise as the ESP32 is on anyhow)
Pushed reset button on the xpro; 
See the reset on fluidterm. Now the boot up goes through OK wo errors.

(I’m online @ Discord as well)

bdring, 2022-04-11:
"See the reset on fluidterm. Now the boot up goes through OK wo errors." 

I think that is the only one you did correctly.

Main power must be on first.

possu911, 2022-04-11:
> Main power must be on first.
I know & this is how I ‘turn’ this on every time.
The other two steps above was just for additional info. Sorry for confusion.

MitchBradley, 2022-04-11:
It would also be interesting to know if $bye from the WebUI console fixes it.

possu911, 2022-04-11:
@bdring 
did following to test:
Modified the "config_new_working" by adding "allow_single_axis: true" to homing sections.
Uploaded the file first w webui. 
Pushed reset on the xpro. 
$SS gave following errors:
[MSG:ERR: X Axis driver test failed. Check connection]
[MSG:ERR: Y Axis driver test failed. Check connection]
[MSG:ERR: Y2 Axis driver test failed. Check connection]
[MSG:ERR: Z Axis driver test failed. Check connection]

Powered xpro off (off switch + unplug USB)
CNC machine is responsive & OK, tested with web ui.
Quit web ui.
fluidterm. Reset button few times, boots up OK every time,

Seems reset button is doing OK.

Still cant change any values from the console e.g. $axes/x/soft_limits=true
[MSG:ERR: Initialization error at /axes/x/motor0/tmc_5160: For daisy-chained TMC, cs_pin: pin must be configured only once]
[MSG:ERR: Initialization error at /axes/y/motor0/tmc_5160: spi_index: must be unique among all daisy-chained TMCs]
[MSG:ERR: Initialization error at /axes/y/motor1/tmc_5160: spi_index: must be unique among all daisy-chained TMCs]
[MSG:ERR: Initialization error at /axes/z/motor0/tmc_5160: spi_index: must be unique among all daisy-chained TMCs]

possu911, 2022-04-11:
@MitchBradley $bye doesnt seem to do anything from webui.

bdring, 2022-04-11:
$bye kills the controller, so the WebUI has to wait until the connection comes back.

possu911, 2022-04-12:
Studied Trinamic TMC5160 datasheet. 
Read about minimum current settings etc. 
Attached picture is from the page with min current calculation example. 
Then looked at Spark-Concept's CNC_xPro_V5_machine_template.h:
They have provided a formula to calculate run_amps and estimate for hold_amps.
Modified my config file accordingly. 
run_amps from 1.8(default) to 1.145
hold_amps from 1.25(default) to 0.858
Unless I'm wrong, these numbers are much higher still compared to calculation w Datasheet (2.0Ohm motors)

Loaded the new file. Got the same errors again.. darn. After power off, usb cable off. boot up OK.
Unless my hearing has gone worse today, I think the steppers run quieter.
Then for sure the Stallguard is working when I run the steppers to halt, previously it was grinding.
i.e some progress.

Schematics, even partial, pertinent to ESP32 & 5160 would help a lot on this.

Thanks
[image]

CNC_xPro_V5_machine_template.h:
//default motor run current
 // this setting = (max motor current * 0.707)*0.9 
 #    define DEFAULT_X_CURRENT 1.8
 ...
 
 //default motor hold current
 //this setting should be roughly 40-75% of the run current
 #    define DEFAULT_X_HOLD_CURRENT 1.25 
 ...

lagnat, 2022-04-12:
> Still cant change any values from the console e.g. $axes/x/soft_limits=true [MSG:ERR: Initialization error at /axes/x/motor0/tmc_5160: For daisy-chained TMC, cs_pin: pin must be configured only [...]

You can change them but it leaves the controller in a broken state.  Save the config with $CD (as I recall) and then reset the controller to load your changes.

possu911, 2022-04-12:
@lagnat 
Yep, that way or modify the config file directly and load the settings. 
Did you get yours to work? Would be good to see your settings and compare (machine differences aside)

MitchBradley, 2022-06-02:
It is unclear whether this issue is still a problem, so I am closing it.  Open a new issue if there is something that is still broken, and reference this one in the new one.

closed 2022-06-02 by MitchBradley