bdring/FluidNC #369: Sender WIFI connectivity

imechura, 2022-04-11:
Is there a solution provided by FluidNC or from elsewhere to allow one to use GCode senders that expect a serial port/com port to connect to grbl to connect to fluidnc over a wifi connection?

I was starting to build a solution but thought I should check to see if something has already been created. If not is this on your road map?

For example, UGS is very popular. As fas as I can tell (Maybe I missed something), UGS only supports connecting to GRBL over TTY or COM port. There are obviously solutions to this but they are all pretty involved and clunky and undocumented from the small amount of research I did on the subject.

pagabee, 2022-04-11:
Hello,
you can connect to fluidnc from UGS through wifi:

https://github.com/winder/Universal-G-Code-Sender/discussions/1833
https://github.com/winder/Universal-G-Code-Sender/wiki/Connecting-the-Controller#smoothie--network

Or from bcnc:
https://github.com/vlachoudis/bCNC/issues/1072

imechura, 2022-04-26:
I have been trying to connect UGS to FluidNC over wifi for over 2 weeks now following the instructions provided in the link above and i have been unable to make it work. Sometimes it appears to connect but even then it is flaky at best. Certainly not something that I would risk running a program on my plasma cutter with.

It feels like not being able to connect a sender over wifi seriously erodes the value proposition of fluidNC. It is the sole reason that I invested the time into converting my machine over from Arduino/Grbl. So now I am left with having put in the time but not actually realizing the benefit. Full of regrets basically.

Is there something I am missing? Some kind of secret documentation that has more than 1 paragraph that shows how to get connected?

I do understand that a lot of this is on the sender projects to implement however it seems like it is in the interest of this project to support this because, what is the point of having wifi connectivity if I can't use it to send and run gcode?

MitchBradley, 2022-04-26:
Wow, with a list of complaints that long, and the sarcastic tone behind them, I just can't wait to spend my uncompensated time helping you.

bdring, 2022-04-27:
Are you serious? 

We don't know anything about your setup. Half of your complaint are about a 3rd party sender. You did not follow any of our methods to ask for help.

Please go back to your Arduino.

closed 2022-04-27 by bdring

imechura, 2022-04-29:
Yes, I am serious and I didn't think my system details are important to respond to the question which is: Is there a way to use fluidnc over wifi with an off the shelf gcode sender.

I tried the suggestions in the original answer I got and they did not work. I am baffled by the notion that this is not a major a concern for you. Is the WiFi connectivity experimental or intended to be used in a limited capacity? If that's the case I did not see that in the documentation.

Is documentation available that describes how a sender would integrate with your system over wifi? The best I can tell is that it requires a HTTP to send commands and WebSockets to receive output but I dont see a way to confirm if that is the recommended approach.

MitchBradley, If you see no value in an honest critique of your project from someone who leverages this technology in a professional context then obviously you and I don't share the same motivation and/or values. I apologize if you find honest feedback offensive.

Maybe you perceive sarcasm but I assure that there is no sarcasm in my message. After putting a significant amount of my own time into trying to make this work and now preparing to scrap it and roll everything back at what appeared to be goal line, I'm not really in the mood for telling jokes.

bdring, 2022-04-29:
The **"Are you serious?"** was a direct quote from your last post. You said it more than once. I thought I would toss that back at you.

We have written more than **1 paragraph** on the wiki. 

The project currently brings in about $3 a day. So we don't feel we owe a lot to people with bad attitudes.

A lot of people use TCP everyday with FluidNC. We have a WebUI that uses Wifi. It works quite well.

Are you using AP or STA mode. STA mode is generally smoother because the ESP32 does not need to do the complete work of an access point.

You can send commands over a simple telnet connection. It uses the exact same protocol that the serial port uses, so a few senders work that way. The port for that is 23.