bdring/FluidNC #370: Wifi do not connect in sat mode after macro rename could not be saved

SnowBeaver, 2022-04-11:
Please only submit bugs for the **latest** version of the code. You can check the version number in the startup messages.

**Do not add the bug label**. The developers of this repo will do that after confirming the issue.

Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

Please answer the following questions.

What version of the firmware are you using?
3.4.3

Is the problem repeatable?
Probably, as I write this issue I am still looking for a way to regain control

Under what conditions does the bug occur?
I tried to rename the macro with a more descriptive name. At first I tried a long 6 letters name, but that failed. However the macro edit window closed and the main webui page was displayed. Macro name was displayed as before naming attempt.  I tried a second time but I limit the macro mane to two letters, I tried to saved but I received an error message. I tried to close the edit window but an error message was displayed. In short I could not get out of the macro edit window. I repowered fluidnc several time  and I could no longer enter the main webui using fluinc.local. I connected with fluidterm and noticed that the wifi could no longer connect in sta mode and revert to ap mode.

Still investigating...

**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```C++
  #define EASIER_TO_READ true
```

SnowBeaver, 2022-04-11:
First attempt I tried the macro name Pshort, then Ps on the second attempt.

bdring, 2022-04-11:
Send $LocalFs/List and paste the reply.

I would like to see what you have on the local file system.

SnowBeaver, 2022-04-11:
Forgot to mentioned that I am using the webui version Mitch suggested with the release of fluidnc 3.4.3

SnowBeaver, 2022-04-11:
I have a job running. I will try in one hour and let you know. BTW I like the webui you and Barbour released.

SnowBeaver, 2022-04-11:
I tried again with different combination of lengths number of macro to edit simultaneously, add delete macro without problem. I guess there is no problem with the macro itself.

closed 2022-04-11 by SnowBeaver