bdring/FluidNC #372: Solenoid not hold below zero

kjk25, 2022-04-12:
try to use solenoid for Z ,

i have a LPKFC30 Milling machine with MKS DLC32,
when i add solenoid to the config like below, the solenoid push for a short time if its below 0.
in both directions, that mean from 1 to -1  Z1 -> Z-1 only a push and nothing else it will not hold the voltage until it get >Z0.
And from Z-1 -> Z1 a push also then nothing

when i move from Z-10 -> Z-5 one push , then Z-5 -> Z-1 one push again.
whats wrong ? who knows the trick ?
best regards

  z:
    steps_per_mm: 1000.0
    max_rate_mm_per_min: 1200.0
    acceleration_mm_per_sec2: 500.0
    max_travel_mm: 80.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      solenoid:
        output_pin: gpio.32
        pwm_hz: 5000
        off_percent: 0.000
        pull_percent: 100.000
        hold_percent: 100.000
        pull_ms: 0
        direction_invert: true

[MSG:INFO: Restarting]
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
[MSG:INFO: FluidNC v3.4.3]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:kjkconfig.yaml]
[MSG:INFO: Machine KJK OnEh]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,325.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.36]
[MSG:INFO: Axis Y (0.000,220.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:     Neg Limit gpio.35]
[MSG:INFO: Axis Z (0.000,80.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     solenoid Pin: gpio.32 Off: 0.000 Hold: 100.000 Pull:100.000 Duration:0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.22]
[MSG:INFO: Connecting to STA SSID:RadioWesterfeldSA]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.5.64]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.4 [FluidNC v3.4.3 (wifi) '$' for help]

bdring, 2022-04-13:
I just tested it and it appears to work for me. It might be due to your **idle_ms** value. If you have a value below 255 all axes will disable, including the solenoid after that duration.

I updated the solenoid section of the wiki with a little more information.

https://github.com/bdring/FluidNC/wiki/FluidNC-Motor-Setup#solenoid

kjk25, 2022-04-15:
worked perfect ! thank you.
would be good to have a idle_ms  for each axes.

best wiches Kai

MitchBradley, 2022-04-15:
> would be good to have a idle_ms for each axes.
That would be a pain to implement.  Getting it right for all axes at once was hard enough.

closed 2022-04-15 by bdring

kjk25, 2022-04-16:
ok i understand, maybe  switch off for a axes would be not so hard ro implement, 
i use the solenoid for Z the motors X,Y are still on also when the mashchine is in standby.
if i could set for a axes idle_x.y.z = false/true
may i am wrong
best regards
kai

bdring, 2022-04-16:
You can disable all or specific motors with the $MD command.

https://github.com/bdring/FluidNC/wiki/FluidNC-Commands-and-Settings

bdring, 2022-04-17:
We discussed this on the Discord dev channel and decided to have the solenoid ignore the idle disable feature.

https://github.com/bdring/FluidNC/pull/389

It will be included in the next release.