bdring/FluidNC #378: RE: Hello and Getting the Discord Invite

silver2row, 2022-04-14:
Hello,

I use 32-bit processors still with beagleboard.org family of boards, e.g. like the BeagleBone Black and so on. Anyway, I always tried to put similar software on it to handle specifics with certain daughter cards people produce which are called Capes. 

Anyways...do you think this FluidNC repo might work with such a board?

Seth

P.S. I am in to 3D Printing, CNC, and motors in general. So, if you are willing to give me an invite, that would be nice. I would like to try to port this to the am335x if possible. If not, I understand.

MitchBradley, 2022-04-14:
FluidNC currently only works on ESP32 processors.  It has many dependencies on ESP32 peripherals and ESP32 SDK features.  In principle it would be possible to port to a different processor, but we currently have no plans to do so because developing and supporting the existing system takes all of our time and then some.  Even if someone else were to do the port, we would have to think long and hard about whether to merge the port, because everything in a code base creates support and testing requirements.

silver2row, 2022-04-14:
Yes Sir,

I understand completely. Whelp, I guess I will try the ESP32. This will be a new venture for me. 

Seth

P.S. Thank you for replying.

closed 2022-04-14 by silver2row

MitchBradley, 2022-04-14:
The other advantage of ESP32 compared to Linux-based solutions is that ESP32 boots much faster, does not require periodic file system checking, and does not require separate administration and updating of operating system components.  Yes I realize that it is possible to use the processors from systems like BeagleBone to run bare-metal code without Linux, but doing so opens up a lot of other issues that must be understood and dealt with.

bdring, 2022-04-15:
https://discord.gg/MpwFYgf7

Geekineer, 2023-01-27:
As I commented to Mitch on his youtube site I'm looking for help to get a plasma table running on FluidNC.  I've been down the same road with Linuxcnc, plasmac, qtplasmac etc.   I'd be willing to be part of a group working on this but my expertise is limited pretty much to the hardware, cnc, and plasma side - not so much on the coding/app side.
Dennis