bdring/FluidNC #379: x modem missing on wifi.bat 3.4.3 version

kweilim98, 2022-04-14:
Does anybody has try with the wifi.bat on the 3.4.3 version after uploading needed the window PowerShell to get the wifi password so only can access to the webui? because i cant reach there, the program show x modem missing and no definite.

MitchBradley, 2022-04-14:
Show us exactly what you did with screenshots.  We cannot understand what you wrote.

kweilim98, 2022-04-14:
[image]

bdring, 2022-04-14:
On Windows, it is easier to use fluidterm.exe, than fluidterm.py.

MitchBradley, 2022-04-14:
There is no file named "wifi.bat" in fluidnc-v3.4.3-win64.zip .  The closest file to that name is "install-wifi.bat".  install-wifi.bat uses fluidterm.exe which does not use the Python code that you show above.  From the information that you have provided, we cannot determine how you got to that Python code.  The screenshot that you provided shows us an excerpt from the fluidterm.py file, but it does not show us what you did to try and run that file.  It should not be necessary to use fluidterm.py on a Windows system.

closed 2022-04-15 by bdring