bdring/FluidNC #380: GPIOPin.12 - Boot loader

boboxx, 2022-04-14:
I'm loading FluidNC for the first time on a ESP32 Devkitv1 connected to some DM542 stepper driver. I loaded the precompiled FW an I'm using the basic 3axis_v4.yaml config.

Looking at the ESP32 pin reference, by default GIOpi12 is ok to use but it's tied to the boot loader. The issue I have is that I have tried to set the pin high or low but it's always entering in boot loader mode when I connect the USB to power up the board if the cable is connected to the the stepper driver or if GPIO12 is connected.

How can I fix this?

  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    homing:
      cycle: 2
      mpos_mm: 10
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.4:low:pu
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12:low
    motor1:
      null_motor:

boboxx, 2022-04-14:
yes that's the reference I looked at:
GPIO_NUM_12
[If driven High, flash voltage (VDD_SDIO) is 1.8V not default 3.3V. Has internal pull-down, so unconnected = Low = 3.3V. May prevent flashing and/or booting if 3.3V flash is used and this pin is pulled high, causing the flash to brownout. See the ESP32 datasheet for more details](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection)
This is a strapping pin feature. You can force the voltage by setting the one time programmable [efuse](https://docs.espressif.com/projects/esptool/en/latest/esp32s2/espefuse/index.html). python espefuse.py --port COM4 set_flash_voltage 3.3V. This will prevent the pin from being checked.

I also tried setting it to this but it was also entering into the boot loader:
motor0:
  limit_all_pin: gpio.4:low:pu
  stepstick:
    direction_pin: gpio.14
    step_pin: gpio.12:high
motor1:
  null_motor:

bdring, 2022-04-14:
What is driving the pin high?

I would not mees with the efuses. Many people have bricked there ESP32 playing with those.

boboxx, 2022-04-14:
I have connected gpio12 to pul-, but pul+ is connected to 3.3v. I followed the diagram on page 9 https://kitaez-cnc.com/f/dm542.pdf

I'm don't really understand why the pin goes high on bootup.

MitchBradley, 2022-04-14:
The stepper input has a optoisolator.  Electrically that looks like a IR LED and resistor in series, connected from 3.3V (PUL+) to GPIO 12 (PUL-).  That is enough to put a voltage on GPIO 12 that it interprets as electrical high, thus triggering the boot problem behavior.
[image]

The efuse solution will work at the risk of making a mistake and causing other problems.
Another solution is to reassign pins so that GPIO.12 is used for some other function whose input does not act like a pullup from an electrical standpoint.
Yet another solution is to connect PUL- to GND and GPIO.12 to PUL+ as shown in figure 4 of the PDF you cite.

closed 2022-04-14 by MitchBradley

bdring, 2022-04-14:
Opts can be connected to the controller on the high or low side.