bdring/FluidNC #383: max_rate

us6iae, 2022-04-15:
Why doesn't max_rate_mm_per_min: 2000 limit machine speed to 2000mm/min? When running Gcode, the speed is taken from the F.

MitchBradley, 2022-04-15:
The target speed in the direction of travel is taken from the F word, and the aggregate speed is limited so that no axis travels faster than the max rate for that axis.  When traveling along a diagonal, the aggregate speed can exceed the speed of any individual axis according to trignometry.

us6iae, 2022-04-15:
And I also found a lot of mistakes.
It shouldn't be.
You check and then close the question.
Sincerely, Alexander.

bdring, 2022-04-15:
The F value sets the desired speed for the move in the direction of travel. Each axis has a max speed. If the value of F and the direction of travel would exceed any max speed the actual speed of the move will be limited so no max speed on an axis is exceeded.

Only when you are doing a move that involves only one axis, will F max equal the max_speed of that axis.

Acceleration works similarly. If you have a very high acceleration on one axis the actual acceleration may be limited if the move also involves an axis with a slower acceleration.

If you have an example where an axis was forced to move faster, it its direction of travel, than its max speed, please give it. Be sure to do the maths on the feed rate and angle of motion to determine what the rates on each axis are. 

If you have found other mistakes, please share them.

us6iae, 2022-04-17:
Good afternoon.
It is difficult for me to write to you. My language is Ukrainian/Russian. I
am writing through google translator. I have been building CNC for over 20
years. Mknya now has 5 machines under different autonomous controllers.
There is something to compare. Tested on different machines. For example -
on a machine with a maximum speed of 4700cm / min - X0 X1000F4500 works
fine on your controller. And if I run Gcode, it doesn't work.
max_rate_mm_per_min: 4000 Speed sometimes increases to 6500cm/min.
max_rate_mm_per_min: 3000 doesn't work
max_rate_mm_per_min: 2500 doesn't work
max_rate_mm_per_min: 2000 - works, but the processing time is almost
doubled.
I am sending you two files, you can check it on a test board even without
motors.
Everything works on all my machines with other control units.
Sincerely, Alexander.

MitchBradley, 2022-04-17:
Please send the files that you mentioned.  Also send the config file (.yaml) for your machine, a picture of your controller, a picture of your machine, and a description of the machine's drive characteristics - type of motors (size, steps/rev), stepper driver and its settings (microsteps), and transmission (ball screws and pitch, pulleys and their number of teeth, or rack and pinion with tooth characteristics).

us6iae, 2022-04-17:
I sent you two files for you to check. You don't even want to check. This
is your right as an author. I will not insist. On the next error, you write
that everything is correct. Write that the project is stopped, and I will
no longer write.
Sincerely, Alexander.

MitchBradley, 2022-04-17:
I really did look for the files but I did not find them.  I do not see any attachments on this ticket.  Please tell me how you sent them.  One way to show that would be to attach a screenshot illustrating the place where the files appear.

bdring, 2022-04-17:
These appear to be email replies rather than sent directly via the Github website.

us6iae, 2022-04-17:
Still does not allow two or more connections to connect. One turns on, the
other turns off. Maybe it was intended that way, but at the moment of
connection, all work stops for 0.5 seconds. And there are skipping steps.
This is one of the big mistakes. And the web really interferes with the
controller. I can take a vmdeo, but it seems you are not interested.
Sincerely, Alexander.

us6iae, 2022-04-17:
I was replying to the mail because you closed the issue.
Sincerely, Alexander.

MitchBradley, 2022-04-17:
I closed the issue because I thought that the problem was solved.  It is very common for people to be confused by the fact that, on diagonal moves, the speed along the hypotenuse of the triangle exceeds the speed of either axis, so the reported speed during a diagonal move can be faster than the max speed configured for an axis.  Your original question was so brief that I thought that was the problem and that we had answered it, as we have already done many times.

Bart, if you have the emailed files, please send them to me or attach them here.  If I received them, they are buried somewhere.

MitchBradley, 2022-04-17:
I reopened the issue.  Neither Bart nor I can find the files that you cited, neither here nor in emails.  Please attach them directly to a comment on this issue - via GitHub, not via email - and also supply the machine information that I cited above.  Please do not repeat the previous conversation in your comments.

us6iae, 2022-04-17:
[4axis_v1.zip](https://github.com/bdring/FluidNC/files/8502193/4axis_v1.zip)
[Uploading kub58mm.zip…]()

us6iae, 2022-04-17:
[kub58mm.zip](https://github.com/bdring/FluidNC/files/8502196/kub58mm.zip)

MitchBradley, 2022-04-17:
Do you have external drivers?  The pulse_us value 2 is faster than some external drivers can handle.  Try increasing it to 4.

us6iae, 2022-04-17:
I tried both 3 and 4 and 5. The drivers are good, external.
There are converters 3v - 5v.

MitchBradley, 2022-04-17:
What kind of converters?  Bidirectional converters do not work for this purpose.

us6iae, 2022-04-17:
Why are there bidirectional ones? I am very good at circuit engineering.
[image]
max_rate_mm_per_min: 2000

MitchBradley, 2022-04-17:
Some people have tried to use bidirectional level translators that are intended for interfacing between CMOS chips for the purpose of interfacing to optocoupled external drivers.  That does not work.  I had no way of knowing that you have circuit design skill before you told me, so I cited a common problem.

MitchBradley, 2022-04-17:
Is the problem that the display shows 2624 mm/min, while the X, Y, and Z axis max rates are 2000 mm/min?

MitchBradley, 2022-04-17:
The max rate values are per-axis.  When moving both axes simultaneously along a diagonal, the vector speed will exceed the speed in the direction of either axis.  The formula, from basis geometry is vector_speed = sqrt(x_speed*x_speed + y_speed*y_speed).  In the case where the direction of motion is 45 degrees and thus x_speed = y_speed, the vector speed is 1.414=sqrt(2) times either individual speed.  It is obvious from the visualizer display that the tool is moving on a diagonal, so it is to be expected that the vector speed is greater than the per-axis limit.
[image]
[image]

MitchBradley, 2022-04-17:
> Still does not allow two or more connections to connect. One turns on, the other turns off.

That is intentional.  The ESP32 chip is not fast enough, and does not have enough memory, to easily manage more than one web connection while also doing motion control at speed.  With a great deal of effort it might be possible to make multiple web connections work, but most of our users prefer that we use our limited development time on other features that fit more easily within the ESP32's constraints.

> Maybe it was intended that way, but at the moment of connection, all work stops for 0.5 seconds. And there are skipping steps.

The reason why work stops when you make a connection is because the ESP32 is sending the web page data to the browser during that time.  The web page data is stored in the ESP32's FLASH memory.  That same memory is used for fetching the CPU instructions.  During the time that the ESP32 is accessing the web page data to send to the browser, it must pause its execution of CPU instructions - except for the limited set of instructions that are stored in the small amount of special IRAM memory.

One way to work around that problem would be to store the web page in some other location, such as on SD card or somewhere on the computer that is running the browser.  Each of those workarounds has other problems, like ease of use and difficulty of administration.  The web interface was developed by another person as a separate project.  His design uses the ESP32's internal FLASH for the web page storage.  We might someday choose to store the web page somewhere else, but it will require substantial coding work on code that is not ours.

So these things that you call "mistakes" are mostly inherent limitations of a microprocessor that costs $4.  It has a lot of capabilities for that price, but its resources are finite and we must be careful how we allocate them.

us6iae, 2022-04-18:
Look closely at the code and show me the movement of the two axes diagonally in the code. There are no such movements. At the end of processing there is a little processing on the circumference, but then the speed does not increase.
Sincerely, Alexander.

MitchBradley, 2022-04-18:
Lines 4961 and 4962:
```
X37.166Y33.254Z-6.851
X36.825Y33.856Z-6.845
```
Delta X is -0.341, delta Y is 0.602.  That is a diagonal move with simultaneous motion in both X and Y.  That is not the only such move; they are all over the place.  Furthermore, with moves that are this short, the planner is likely to coalesce segments that are almost collinear.

The reason that I chose that particular one is because it is very close to the coordinates of the picture above.

MitchBradley, 2022-04-18:
There are 4 places in every loop of the circular pattern where the motion is close to a 45 degree angle, and thus delta X and delta Y have nearly the same magnitude.  For example:
Line 33316:
```
X9.750Y50.470
X8.925Y49.702
```
Delta X is 0.795 and delta Y is 0.8.  At such points the instantaneous speed will reach sqrt(2) * per_axis_max, or 2818 - but it is unlikely that you will see that number on the display unless the reporting sample happens to be take at just the right time.

The way you are using GCode and controller max rates is not ideal.  Your GCode has the feed rate set to 10000 and the controller then limits the per axis rates to a lower speed.  That will result in different surface speeds at different points in the circle.  The conventional way is to set the GCode feed rate to a speed lower than any per-axis maximum. That will result in a consistent surface speed regardless of the direction of cut.  The controller will not have to enforce a limit during a cutting motion.  The limits would apply only to rapid motion and would provide some safeguards against mistakes.

us6iae, 2022-04-20:
You forgot to add acceleration in the formula. And if you think that a stepper motor is capable of reaching a speed of 2818cm / min, then I built a spaceship. And every time this moment hits the screen.
And I have a file with codes that is universal for different machines.
Close the question. We spent too much time discussing. There are still issues that need to be addressed.

Sincerely, Alexander.

MitchBradley, 2022-04-20:
This is the way that GRBL has worked since forever.

closed 2022-04-20 by MitchBradley