bdring/FluidNC #387: Machine is not homing

Mamoonpk, 2022-04-15:
Version 3.4.3. Win64
Board 6 Pack with external drivers

Need help on homing, I can jog the machine using Web UI but homing is not working. the moment I go for homing it triggers a hard limit switch false alarm.

I have done the steps mentioned in https://github.com/bdring/FluidNC/wiki/Help-with-Switch-Problems
but on the final step $HX it gives the error Alarm1. My spindle is in the middle and all the hard limit switches are clear.

I am attaching the machine configuration along with the log of each step mentioned in help-with-switch-problems for homing.

please advise. 

[6_Pack_External_XYZY2.txt](https://github.com/bdring/FluidNC/files/8497552/6_Pack_External_XYZY2.txt)

Please only request help for the **latest** version of the code. You can check the version number in the startup messages.

Please read the entire [wiki section about requesting help](https://github.com/bdring/FluidNC/wiki/Requesting-Help).

Please read the entire [FAQ section](https://github.com/bdring/FluidNC/wiki/FluidNC-FAQ).

Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

What version of the firmware are you using?
[fluidterm log.docx](https://github.com/bdring/FluidNC/files/8497567/fluidterm.log.docx)

Insert boot messages below.

Using COM3
Ctrl-] to exit, Ctrl-U to upload, Ctrl-R to reset
Resetting MCU
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8

[MSG:INFO: FluidNC v3.4.3]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:6_Pack_External_XYZY2.yaml]
[MSG:WARN: gpio.39 does not support :pu attribute]
[MSG:WARN: gpio.36 does not support :pu attribute]
[MSG:INFO: Machine 6 Pack External XYZ]
[MSG:INFO: Board 6-pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:5us Dir Delay:6us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2:low Dir:I2SO.1:low Disable:I2SO.0]
[MSG:INFO:     All Limit gpio.33:low]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5:low Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:     All Limit gpio.35:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.13:low Dir:I2SO.12:low Disable:I2SO.15]
[MSG:INFO:     All Limit gpio.34:low]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10:low Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:     All Limit gpio.32:low]
[MSG:INFO: Reset gpio.2:pu]
[MSG:INFO: FeedHold gpio.25:pu]
[MSG:INFO: CycleStart gpio.39]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Probe Pin: gpio.36]
[MSG:INFO: Connecting to STA SSID:TP-Link_MR2.4G]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.30.140]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.3 (wifi) '$' for help]
[MSG:INFO: Check limits]
[MSG:INFO: '$H'|'$X' to unlock]

Avataar120, 2022-04-15:
Do you have pullups on your limit inputs?
If no, you can activate them in the config file adding :pu after the pin definition

Message check limits makes me think limits are either working the other way around or you are missing pullups

MitchBradley, 2022-04-15:
The messages show some problems:
```
[MSG:WARN: gpio.39 does not support :pu attribute]
[MSG:WARN: gpio.36 does not support :pu attribute]
```
Those messages mean that you have asked to turn on internal pullups on GPIOs 39 and 36, but the ESP32 chip does not have internal pullups on those GPIOs, so it is not possible to turn them on.
```
[MSG:INFO: Check limits]
```
That means that one or more limit switches is already active when the system starts.  The most likely reason is that you have the wrong logic sense for some of the switches.  You might need to remove the ":low" designation on some of the "limit_all_pin:" lines.

If you issue the "?" command it will show which pins are currently active in the "Pn:" field.

Mamoonpk, 2022-04-15:
After getting the  Alarm 1, I entered command ? 
below is the log, 

further, I am using 2 motors on y axes, and both motors have separate limit switches on gpio.35 and gpio.34

I have removed GPIOs 39 and 36 and assigned gpio.25 to the probe. however it was working fine with the same setting on ESP32_GRBL. New config file is attached, 
[6_Pack_External_XYZY2.txt](https://github.com/bdring/FluidNC/files/8497989/6_Pack_External_XYZY2.txt)
.000|FS:0,0|Pn:XYZ|WCO:0.000,0.000,0.000>

[MSG:INFO: Caution: Unlocked]
ok
$hx
ok
ALARM:1
[MSG:INFO: Reset to continue]
<Alarm|MPos:0.000,0.000,0
[6_Pack_External_XYZY2.txt](https://github.com/bdring/FluidNC/files/8497989/6_Pack_External_XYZY2.txt)
.000|FS:0,0|Pn:XYZ|WCO:0.000,0.000,0.000>

MitchBradley, 2022-04-15:
Pn:XYZ means that all switches are reporting as active.  I see that you did not remove :low from the limit_all_pin: lines like I suggested.  What is the point of asking questions if you will not try the suggested fix?

bdring, 2022-04-15:
You need to figure out why the switches are reporting as active.

Try running this test and pasting the results here. Do it with Fluidterm rather than the WebUI.

https://github.com/bdring/FluidNC/wiki/FluidNC-Homing#testing

bdring, 2022-04-15:
Can you send a photo of your controller?

Mamoonpk, 2022-04-15:
[image]

The picture of the board is attached.
thanks, I am able to home now after following mitch instructions.

however, I have removed GPIOs 39 and 36 for now from configuration. I want to know that is there a way to use these GPIOs. I want to use it for hold and resume.  I have not connected the Hold, Resume, reset and Probe for now but I wanted to configure them and have plan to setup shortly. 

secondly. I am trying to upload gcode file to SD card using WebUI. the card is already inserted in board. is there something missing in config?

[image]

Mamoonpk, 2022-04-15:
Config File is attached
[6_Pack_External_XYZY2.txt](https://github.com/bdring/FluidNC/files/8498387/6_Pack_External_XYZY2.txt)

bdring, 2022-04-15:
You should be able to use 36 and 39 for normally open switches with that hardware with 

```yaml
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36
  cycle_start_pin: gpio.39
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
```

Note: Ideally you should be using .yaml or .yml for your file extension. Some editors work better when it knows the type.

Try reading the SD first by sending **$SD/List** You may need to play with the format to get it to work.

https://github.com/bdring/FluidNC/wiki/SD-Card

MitchBradley, 2022-04-15:
See especially this section: https://github.com/bdring/FluidNC/wiki/SD-Card#card-formatting

New high-capacity SD card often do not work with the Arduino library that we use.  Older, smaller cards with FAT32 filesystems work better.

Mamoonpk, 2022-04-16:
Mitch and Barton, Thanks for Help. I am all set now. 
SD card formatter worked for me. Homing is also fine now.
I will be running a few protects and will update if I encounter any issues.

Thanks
Mamoon

closed 2022-04-16 by MitchBradley