bdring/FluidNC #388: Spindle PWM wraps in ceratain conditions

kartom, 2022-04-16:
Choosing the following setup causes the PWM signal to wrap when speed is over 50%:
pwm_hz: 600 (or lower? tested with 500 with the same result)
speed_map: 0=0.000% 24000=100.000%

This setup gives an incorrect relation betweein speed and PWM ratio:

s2400: 10%
s6000: 25%
s9600: 40%
s12000: 50%
s14400: 10%
s18000: 25%
s24000: 0% (or 100%, it was constant but i missed to check if it was high or low)

Changing the pwm_hz to 700 (or higher? tested with 800 with the same result) gives a correct result:

s2400: 10%
s6000: 25%
s9600: 40%
s12000: 50%
s14400: 60%
s18000: 75%
s24000: 100% (this time i checked the level)

CastIrony, 2022-05-04:
I just noticed the same issue. My spindle controller manual suggested a using the really low PWM frequency of 50Hz, and the duty cycle output was completely wrong across the board (verified with an oscilloscope). 

I Increased the frequency back to the default of 5000Hz and the output worked perfectly, and my controller didn't seem to mind, so I didn't do extensive testing to determine the exact cutoff. It being somewhere between 500 and 700Hz as @kartom suggested sounds plausible.

bdring, 2022-05-04:
I can reproduce the problem. There is an overflow in the duty when the resolution is higher than 16 bits. A fix will probably be in the next release.

closed 2022-06-02 by bdring