bdring/FluidNC #391: Can a 74AHCT595 IC drives 8 optocouplers correctly?

pagabee, 2022-04-20:
I want to plan a CNC board and have a question as a hobbyist...

The 74AHCT595 IC's specification says:
Ignd max = -75mA
Icc max = 75mA
Io max = +/-25mA

The signal current of the DM542 or the DM556 stepper drivers are between 7-16 mA.
The average current is 11.5mA, that means 8 × 11.5mA = 92mA, that is higher than the Icc max 75mA.
My chinese drivers (TB67S109) need 16,6mA at 5V.
Would it be more advisable to use driver IC-s after the 595 IC?
Forexample: ULN2803 or SN74LVC2T45?
As hobbyist I would prefer to use the ULN2803, because it is cheaper, easier to buy and solder.

What are your suggestions?
Thanks in advance!

bdring, 2022-04-20:
I would not driver that much current through one 595. I do not know what sets the limit. One might argue that step pulses are such a low duty cycle they would not stress the part, but I would still avoid it.

An easy solution might be to add a second 595 and use less channels per part. The 6 pack basically does that when used with external drivers. You might be able to use those extra channels for other things in the future.

MitchBradley, 2022-04-20:
You can add external series resistors to drop the current down to the 9 mA range.  The drivers have internal current limiting resistors whose values are a compromise intended to allow operation with a variety of input voltages.  It is extremely likely that the input optocouplers will operate correctly with currents down to 5 mA and often quite a bit lower than that.  The optocoupler current is (Vin - Vf) / R .  Vf is very close to 1.2V for typical operating conditions.  If Vin is 5V, then the current is (5-1.2)/R = 3.8/R.  The R resistor is inside the driver box on the circuit board.  Sometimes its value is 270 ohms but that is not guaranteed.  You can add an additional external resistor in series to increase the value and drop the current to the value that you want.  One way to determine the appropriate external resistor value is to drive the input with 5V from a power supply and measure the current with a ammeter.  Then either use the calculation above to choose additional resistance to add in series - or just swap in some resistors until you find one that gives you the desired current.  If the internal resistor is the typical value of 270 ohms, then an additional 270 ohm resistor in series will drop the current to about 7 mA.

Another approach is to drive the input at 3.3V and measure the current. If the current is about 7mA with 3.3V drive, you could just use 3.3V as the VCC for the '595s.  Even if the current is a bit lower, say in the 5mA range, it is very likely that the drivers will still work.  Most optocouplers are *much* more sensitive than their guaranteed specs.  The guaranteed specs are usually extremely conservative, rated across temperature extremes and years of aging at the high end of the allowable input current.  When operated at reasonable temperatures at moderate input currents, they will exceed their specs for decades.

pagabee, 2022-04-21:
Thanks for the suggestions!

I do a little research on the Internet...
Many CNC boards use 74HC245 as output driver. This IC has very similar I/O current rate as the 74AHCT595.
The boards are working, but I don't know the reliability of them, how correct can they drive the stepper drivers.
I will try the series resistor.
[image]
[image]

MitchBradley, 2022-04-21:
Closing because there is really nothing more to say on this.

closed 2022-04-21 by MitchBradley