bdring/FluidNC #393: Stepper movement stops when running a ? command

haraldkubota, 2022-04-23:
Please only submit bugs for the **latest** version of the code. You can check the version number in the startup messages.

**Do not add the bug label**. The developers of this repo will do that after confirming the issue.

Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

Please answer the following questions.

What version of the firmware are you using? 3.4.3

Is the problem repeatable? Yes

Under what conditions does the bug occur?

I send 4 commands via TCP connection (telnet server) to move the x axis: move to 0, then to 100, then to 0, then to 100. This works flawlessly and smooth.
I then added to send a "?\n" every 200ms to check if the controller is idle, but when I send or receive the result, the motor seems to stop. See https://youtu.be/pRsrsQokiOg
Changing the 200ms to 1s instead, I get corresponding less "jerky moves". They also happen at exactly the same point in time. Very reproducible, so it seems to be that the processing of incoming traffic causes the stepper loop to halt for a short time.
Same I can do manually: Send a long move (G0X0 then G0X400), and during the move send ? to get the current position. You'll get a very short but noticeable stop.

haraldkubota, 2022-04-23:
Forgot add: Using MKS DLC32 board. Config for the x axis:
```
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 428.0
    max_rate_mm_per_min: 15000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 440.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 5000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      hard_limits: true
      pulloff_mm: 2.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
```

MitchBradley, 2022-04-23:
428 steps/mm * 15000 mm/min / 60 sec/min is 107000 steps/sec which is very close to the maximum I2S step rate of 125000 steps/sec.  You are using I2S_STATIC which requires a lot of software involvement in the step generation.  It is not at all surprising that motion starts to be troublesome when the system must do additional work to send responses back over the network.  You are trying to run the system faster than the processor can handle reliably.

haraldkubota, 2022-04-23:
Thanks for the quick response. I reduced the microsteps and have now 40.7 steps/mm which is now about 10k steps/sec. I also use I2S_STREAM now to reduce CPU usage. It's much better now and many, but not all ? commands cause no pulse interruption. It's less reproducible though which makes debugging a bit harder.

haraldkubota, 2022-04-23:
For fun I used I2S_STATIC and while I had to use 53.5 steps/mm, I now again get short stop-and-go when I do a ? in the middle of a move. 
I2S_STREAM is better, but the problem is not gone even with those reduced step rates.

MitchBradley, 2022-04-23:
I don't know. We have not seen the situation where I2S_STATIC and I2S_STREAM require different steps/mm. That suggests to me that the step timing or pulse polarity might be wrong for you stepper drivers.  It is hard to debug system level problems with limited knowledge of the overall system.

haraldkubota, 2022-04-23:
Ok, this is all new to me but knowing that the numbers should match for I2S_STATIC and I2S_STREAM means I can debug a bit and maybe find out how to fix this "interrupt of movement when doing a ?". Thanks. I will close this ticket since the indication is that something is "wrong", possibly with my hardware.

closed 2022-04-23 by haraldkubota

MitchBradley, 2022-04-24:
Also you should know that rapid-fire status over TCP is not something that we have tested very much.