bdring/FluidNC #394: why only 1 step jog motion is not posible ?

MAK-1966, 2022-04-24:
hello dear forum,

I am trying to move the step motor 1 step with the $J= X0.001---- jog command
however grbl is accumulating the jog commands
and moves 0.010 when the total of 1 step jog commands is 10

I want to move 1 step exact (0.001) with jog command

is this not posible ?

thank you

bdring, 2022-04-24:
Show your config for that axis.

MAK-1966, 2022-04-24:
hi, thank you
step driver steps/rev is 2000 (microstep dipswitch 10) and $100 attached I am using X axis at the moment

[image]

bdring, 2022-04-24:
#1 That text did not come from FluidNC. We do not use $ settings. We use a config file.

#2 With Grbl firmwares that do support $ setting. $100 is the X resolution. It is set to 100 steps per mm. This means the minimum motion is 0.010.

The setting on your driver is only the microstep setting and is only one part of the resolution calculation for a machine. Pulleys, Leadscrews, etc must also be included.

0.001mm is a very small step. This can be achieved, but you will severely limit how fast the machine can go duw to the high step rate.

MAK-1966, 2022-04-24:
thank you
I try to eliminate PC and add PLC function to GRBL 
this is my first try with Arduino this time 
I hope next I can add fluidNC to my project
I am following FluidNC with attention
[image]

MitchBradley, 2022-04-24:
Closing since it is not a question about FluidNC specifically.

closed 2022-04-24 by MitchBradley