bdring/FluidNC #395: Development of a new raster protocol

arkypita, 2022-04-26:
Hello, I'm the developer of LaserGRBL.

The gcode and grbl are great, but the use made of them for raster image engraving is a strain on its capabilities (as well as being highly inefficient).

For this reason I would like to try to write the specifications of a protocol for sending raster data, something that involves sending some modal commands at the beginning (to program the X / Y step between one point and another, the intensity desired for black, the dwell time on each point etc) followed by a series of data that define only the gray values of each point, ​​and a line change command.

I intend to do it in two possible ways:
- fixed point time, modulated power
- fixed power, modulated point time

All obviously taking into account the length of the buffer, the management of immediate commands (therefore using escape codes) etc.

**Here is my requests**
1) do you know if something like this already exists? I would not want to redo a job that already exists!
2) do you know if there is an official list of existing gcodes (and free ones) so I can choose gcodes that do not conflict with other features?

I know that in the standard grbl there is no chip space for other features, but I think it can be developed on other platforms like ESP32/FluidNC. Are you interested on it?

Anyone who wants to collaborate can write under this thread: https://github.com/arkypita/LaserGRBL/discussions/1818

MitchBradley, 2022-04-26:
I am interested.  I am a core developer for FluidNC. I have a lot of experience in developing computer industry specifications, having been the Vice Chair of the ANS committee that developed the standard for the FORTH programming language and the Chair of the IEEE committee that developed the Open Firmware standard.  I have had laser raster mode discussions before with the TinyG/g2core developers.  I will switch over to the thread that you cite above.

bdring, 2022-04-26:
I am interested as well. I have experience with non gcode based laser protocols.