bdring/FluidNC #398: Laser pwm latency

yu7tom, 2022-04-28:
I have a problem with laser enable pin and pwm pin.They dont mach up.
In LightBurn i drew 1mm wide line and when i engrave it at 200mm/s there is 30ms delay in pwm and enable.
Is there a way to lower that delay? 
LIghtBurn dose not use M5 to turn of laser, it is done with s0.
My plan was to hold S value with a voltage peak detector and to fire the laser with enable pin.
This is the g code i ran:

G00 G17 G40 G21 G54
G91
M4
; Scan @ 100 mm/sec, 50% power
M8
G0X20Y100.77
; Layer C00
G1X-20F6000S0
G1X-1S500
G1X-20S0
G1Y-0.07S0
G1X20S0
G1X1S500
G1X20S0
G1Y-0.07S0
G1X-20S0...

On the scope ch1 is enable_pin: gpio.27 and ch2 is output_pin: gpio.16  pwm_hz: 22000

[image]
[image]
[image]

MitchBradley, 2022-04-28:
Thanks for providing good data to help us understand your problem.

The root cause is the fact that the enable pin is synchronized to GCode parsing while the PWM pin is adjusted dynamically during stepping.  There is a delay between parsing and stepping because the parser sends line segments to the planner, which computes a motion plan that splices together multiple segments, applying acceleration/deceleration constraints to the segment join points, which can include direction changes that affect acceleration.  That must be done across groups of segments in order to get the speeds right at the endpoints and corners.  The planner then feeds the motion plan to a step generator that implements the plan.  In order to achieve smooth motion, there are multiple levels of queuing in this chain.  Otherwise the system would come to a halt while waiting for the next information to arrive.  That queueing creates the latency that you have observed.  Reducing it would result in stuttered motion.

The S value is propagated through all the queues so that it can be applied dynamically, synchronized with the step generation and power-modulated according to motion speed.  The enable value is not propagated in that way, but rather is applied while motion is stopped.

GCode - and Grbl - was originally designed for controlling things like milling machines with rotating spindles.  The M3..M5 behavior reflects that assumption.  Laser mode was later shoehorned into Grbl, but neither GCode nor Grbl are really designed for laser use at the deepest level.  The GCode language itself is ill-suited for the needs of laser engraving, and the Grbl line protocol has additional problems that make it a poor fit for many laser tasks.  In some cases it can be made to work, but it is sort of like trying to use a screwdriver when you really want a proper chisel.

There is an [ongoing discussion on the LaserGRBL repo](https://github.com/arkypita/LaserGRBL/discussions/1818) about ways to improve the situation, but so far there is no good consensus or plan.

yu7tom, 2022-04-28:
Thanks for the quick reply. I figured the answer wouldn't be simple.
I fed the input to an op amp whit high gain and got what i neded.
The response time of a co2 laser psu is 1ms so anything below that should be ok.
I will post my results once i test this on a laser machine.
I have a few proper chisels but its always fun to improve a screwdriver .
[image]

closed 2022-04-28 by MitchBradley