bdring/FluidNC #399: Wifi Connectivity Continued

imechura, 2022-04-29:
Yes, I am serious and I didn't think my system details are important to respond to the question which is: Is there a way to use fluidnc over wifi with an off the shelf gcode sender.

I tried the suggestions in the original answer I got and they did not work. I am baffled by the notion that this is not a major a concern for you. Is the WiFi connectivity experimental or intended to be used in a limited capacity? If that's the case I did not see that in the documentation.

Is documentation available that describes how a sender would integrate with your system over wifi? The best I can tell is that it requires a HTTP to send commands and WebSockets to receive output but I dont see a way to confirm if that is the recommended approach.

MitchBradley, If you see no value in an honest critique of your project from someone who leverages this technology in a professional context then obviously you and I don't share the same motivation and/or values. I apologize if you find honest feedback offensive.

Maybe you perceive sarcasm but I assure that there is no sarcasm in my message. After putting a significant amount of my own time into trying to make this work and now preparing to scrap it and roll everything back at what appeared to be goal line, I'm not really in the mood for telling jokes.

haraldkubota, 2022-04-30:
I do not have an "off the shelf gcode sender", but I send G-code commands via WiFi via a TCP (telnet) connection from my own program (adjust some parameters via UI, click "Do it" and off it sends off G-Code via that TCP connection to FluidNC.

Similar to old telephone modems, I do have to initialize my controller once (e.g. sending $H and wait for it to not error out). But that's ok for me.

So what problem do you have?

MitchBradley, 2022-05-02:
Regarding "I didn't think my system details are important": When something is not working - but works for others - details are always important.

"Off the shelf gcode sender" is easy to say but very difficult to pin down.  There are many many GCode senders but only a very few that are actively maintained.  Back in the day, nearly all GRBL controllers ran on AVRs with only serial connections, and were compiled from the same source code, with the only substantial variation being the version, usually either 0.9 or 1.1.

The ability to connect via WiFi or Ethernet appeared only recently, and in different forms depending on the fork or variant of GRBL - grblHAL, Smoothieware, Grbl_Esp32/FluidNC, and probably others.  There is no coordinated spec for how it should work because the developers of those variants are focused on their own interests and do not work together at any significant level (I once reached out the grblHAL developer and proposed some coordination in conjunction with UGS; he wasn't interested).  Since there is no common spec for how network connectivity works, and the majority of GRBLs do not support it anyway, the sender developers do not have a great incentive to spend a lot of time on it.  In fact, many or even most of the existing "off the shelf" senders do not have any active developers.  Developing and supporting software like this, which used to be simple when there was only "one" GRBL and expectations were low, but has now become very complicated, is extremely time consuming and there is virtually no money to be made - but there are thousands of naive users with high expectations for what they should receive without paying anything.

So, "off the shelf gcode sender" is pretty much not a thing when it comes to network connectivity.  There are some specific senders that sort of support network operation and sort of work with specific GRBL (or Smoothieware) variants, but as a broad category, most gcode senders do not support network connectivity.  One specific example is IOSender which is heavily tuned for grblHAL - in fact it is primarily intended to test and showcase grblHAL.  In the FluidNC space, we support ESP3D WebUI and ensure that it works.  UGS is reported to work, but we have no control over its development other than talking to its developer from time to time - but he is usually very busy.

Regarding the recommended way to connect to FluidNC over the network: You can use either a raw TCP connection on the port given by the $Telnet/Port parameter or a WebSocket connection on $HTTP/Port + 1 - usually port 81.  I prefer websockets.  In either case, you can send the standard GRBL line protocol over the resulting stream, with the same "ok/error" flow control that is used over serial line.  The (poorly documented - it was developed by a third party) HTTP API can be used for things like upload and download of files to SD and the local FLASH filesystem.  You can also use it to send individual commands (which can be send over the TCP/websocket streams as well), but the HTTP API is not suitable for streaming GCode like a sender does.

All that said, it is better to run GCode from a local SD card versus streaming ("sending") over the network.  The SD card is not subject to job interruption if the network has problems.  If one had a wired Ethernet connection, that might be a good channel for streaming GCode - indeed it would likely be better than USB serial which is notoriously sensitive to electrical noise from spindles and motors - but ESP32s are optimized for WiFi and rarely have wired Ethernet.

The use case that we spend nearly all of our limited development and support time on is WiFi operation using WebUI, running GCode from SD card.  We have control over both ends of that connection and we can ensure that it works. Some people like to use traditional senders over USB serial, and we support that reasonably well.  A few use one of the very few senders that support some form of networking connection, but so far that has been a very small proportion of users, and since we do not control the development of those senders there is not a lot that we can do to make sure they work flawlessly.  There are also some pendants that use TCP for jogging and job initiation.