bdring/FluidNC #404: FluidNC Firmware Upgrade Issue - Motors not powering on

Piercre, 2022-05-04:
Hello,

I have a machine that was fully operational and decided to update from GRBL_ESP32 to Fluid NC. No hardware/wiring connections were modified or disconnected/reconnected. The firmware update went fairly well using the conversion tool. I have no errors being reported. I am able to communicate with the machine via the WebUI, but none of the motors are being powered on or able to be jogged. I am able to move them freely by hand. Note: I am using a fairly barebones setup (XYYZ) without any other features (i.e. no endstops, spindle controller, probe, etc).

I have tried the following:
- Changing Disable Pin to 'NO_PIN'
- Changing Disable Pin from high to low
- Initializing the motors using $MI
- Rebooting the controller using $Bye

The setup of the firmware is not my cup of tea, so I am at a loss of how to debug further than where I am at. I am on the Discord page and can be tagged anytime with @Viper-III if that is any easier to work through any ideas or questions.

Controller Type: 6 Pack Universal CNC Controller v1p5

[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack StepStick XYYZ Root 3 CNC]
[MSG:INFO: Board 6 Pack Universal CNC Controller Firmware V1p5]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8:low]
[MSG:INFO: Axis Z (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: Connecting to STA SSID:PierceWifi]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.188]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: SSDP Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]

```yaml
board: 6 Pack Universal CNC Controller Firmware V1p5
name: 6 Pack StepStick XYYZ Root 3 CNC
meta:

stepping:
  engine: I2S_stream
  idle_ms: 255

  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 6

axes:
  shared_stepper_disable_pin: NO_PIN
  shared_stepper_reset_pin: NO_PIN
  x:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: false

      positive_direction: true

      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0:low
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.3
        reset_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

  y:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: false
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7:low
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.6
        reset_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8:low
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.11
        reset_pin: NO_PIN

  z:
    steps_per_mm: 100.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 0
      allow_single_axis: false
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 200.000
      seek_mm_per_min: 2000.000
      settle_ms: 250
      seek_scaler: 1.100
      feed_scaler: 5.000

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15:low
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: I2SO.14
        reset_pin: NO_PIN

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      null_motor:

kinematics:
  Cartesian:

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false
  deactivate_parking: true
  check_limits: false

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
NoSpindle:
```

$S
$Firmware/Build=
$Report/Status=1
$Config/Filename=config.yaml
$Message/Level=Info
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Telnet/Enable=ON
$Telnet/Port=23
$HTTP/Enable=ON
$HTTP/Port=80
$WiFi/Mode=STA>AP
$Sta/SSID=PierceWifi
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=FluidNC
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$Hostname=fluidnc
ok

$I
[VER:3.4 FluidNC v3.4.4:]
[OPT:PHS]
[MSG: Machine: 6 Pack StepStick XYYZ Root 3 CNC]
[MSG: Mode=STA:SSID=PierceWifi:Status=Connected:IP=192.168.1.188:MAC=A8-03-2A-6B-82-D0]
ok

?
<Idle|MPos:0.000,0.000,0.000|FS:0,0>

bdring, 2022-05-04:
If you can freely turn the motors by hand, try sending $Motor/Disable or $MD.

This will invert the disable state. If the motors lock, that is a big clue.

Piercre, 2022-05-04:
Unfortunately there was no change with sending $MD.

bdring, 2022-05-04:
What type of stepper drivers do you have?

Most disable pins do not use the the low attribute. 

There are a ton of examples here.

https://github.com/bdring/6-Pack_CNC_Controller/tree/main/FluidNC_configs

Piercre, 2022-05-04:
Using DRV8825 drivers. I started without the ":low" and added that based on the FAQ to try toggling it. It didn't seem to make any difference.

MitchBradley, 2022-05-04:
Just a shot in the dark - is it possible that the motor power is turned off or otherwise is not getting to the stepper drivers?  I had to ask ... If you have a voltmeter it is easy to measure VMot across the end pins on the motor side of the stepstick modules.

MitchBradley, 2022-05-04:
Connect leads to these points.  You could also use the GND next to the VMOT but the one at the end is easier to reach without shorting the probe leads together.  The hand drawn box at the bottom shows the orientation relative to the 6-pack socket - the box is the motor connector.
[image]

Piercre, 2022-05-05:
[image][image][image]

Piercre, 2022-05-05:
All 4 sticks measured ~0.86V at the Vmot pins.

MitchBradley, 2022-05-05:
The green LED next to the power converter circuit in the middle of the board is not on either.  That suggests a power connection problem.  Measure the voltage right on the power connector, and make sure it is plugged in all the way.

MitchBradley, 2022-05-05:
Oh, you already showed the voltage at the power connector.  There is probably a broken or loose wire between the power supply and that connector. Make sure the wire is all the way in the connector, and tight.

bdring, 2022-05-05:
Mitch is correct. That LED right next to @buildlog should be lit. Measure the voltage there.

Piercre, 2022-05-05:
1.08V across the LED. Shouldn't the voltage into the terminal at the board measure the same 12V at the power supply?

Also, the voltage at the disable pin is 3.26V during a jog command. When I do $MD the voltage drops to 0 volts.

MitchBradley, 2022-05-05:
> Shouldn't the voltage into the terminal at the board measure the same 12V at the power supply?

Yes that is what I meant when I suggested a broken or loose wire.

bdring, 2022-05-05:
It looks like you have a wiring problem for the primary voltage. The disable pin looks good.

Try checking the continuity from the p/s terminals to the main terminal block terminals.

Piercre, 2022-05-05:
Wow, that was embarrassing... the terminal screw was just slightly loose on the positive terminal.  I didn't even think about checking that since I never touched a thing and the matching was working fine 5 minutes before updating the firmware!

I do need to revert the pin state back to high since everything thing starts up disabled and $MD enables the motors.

Y'all are rock stars. Thanks again for the helping the less knowledgeable like myself!

bdring, 2022-05-05:
OK. You should probably increase the wire gauge to at least 16AWG while you are fixing it.

Closing...

closed 2022-05-05 by bdring