bdring/FluidNC #405: Mechanical error compensation

dzario, 2022-05-04:
Hi.
I build my own CNC from polymer concrete
[https://photos.app.goo.gl/WLovmMFLqDHvX3r19](https://github.com/Paciente8159/uCNC/issues/url)
and I have some mechanical error , search CNC control
system for software compensation.
Is possible do it with FluidNC ?
3D printer marlin have Skew compensation .. Any idea?

bdring, 2022-05-04:
We only have auto squaring for dual motor axes.

I don't know what you mean by skew. Your link is broken. Is the machine like a non right angle parallelogram? Some compensation might be possible via kinematics, but you would need to write that yourself.

dzario, 2022-05-04:
Hi. Sorry ,now link working .
Yes, parallel , right angle etc .
Marlin have skew function .

bdring, 2022-05-04:
Use Marlin then.

closed 2022-05-05 by bdring