bdring/FluidNC #408: Config file causes reboot

mikey526, 2022-05-07:
Hi,
I'm setting up a Root Controller Rev2.1 to control my QueenBee Pro. The Root Controller uses GPIO pins mostly for inputs, and I2SO for outputs.
For initial testing I had previously used a bare ESP32 controller with a breakout board using GPIO only, and the config file was working fine.
When setting up the Root controller I started with the same config file, but adapted the I/O pins to match the new controller. I changed some of the GPIO pins for I2SO, added the i2so section, and changed the engine from RMT to I2SO_STREAM.
After uploading the config file I get this at boot time:

[MSG:INFO: FluidNC v3.4.4]                                                                                                                                                                                                                                               
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]                                                                                                                                                                                                                  
[MSG:INFO: Configuration file:config.yaml]                                                                                                                                                                                                                               
[MSG:ERR: Validation error at /axes/x/motor0/standard_stepper: Step pin must be a GPIO pin]                                                                                                                                                                              
[MSG:ERR: Validation error at /axes/y/motor0/standard_stepper: Step pin must be a GPIO pin]                                                                                                                                                                              
[MSG:ERR: Validation error at /axes/y/motor1/standard_stepper: Step pin must be a GPIO pin]                                                                                                                                                                              
[MSG:ERR: Validation error at /axes/z/motor0/standard_stepper: Step pin must be a GPIO pin]
[MSG:INFO: Configuration is invalid]
[MSG:INFO: Using default configuration]
abort() was called at PC 0x401ad2f3 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008a224:0x3ffb1bf0 0x4008a4a1:0x3ffb1c10 0x401ad2f3:0x3ffb1c30 0x401add18:0x3ffb1c50 0x401ad1de:0x3ffb1c70 0x4010df26:0x3ffb1ce0 0x4010e0e9:0x3ffb1d20 0x401ac917:0x3ffb1dd0 0x400e4917:0x3ffb1df0 0x400d5e00:0x3ffb1e10 0x400d5a8b:0x3ffb1e30 0x400d5ac1:0x3ffb1e50 0x400db71a:0x3ffb1e70 0x400db74d:0x3ffb1e90 0x400dc6f1:0x3ffb1eb0 0x400dce85:0x3ffb1f30 0x400dd92f:0x3ffb1f60 0x4010d74a:0x3ffb1fb0 0x4008c152:0x3ffb1fd0

Rebooting...

[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Using spindle NoSpindle]
[MSG:INFO: AP SSID FluidNC_E8CB6F40 IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]
[MSG:INFO: '$H'|'$X' to unlock]
ok

My config:

$Localfs/Show=config.yaml                                                                                                                                                                                                                                                

board: "Root Controller 2.1"                                                                                                                                                                                                                                             
name: "Custom QueenBee Pro"                                                                                                                                                                                                                                              

junction_deviation_mm: 0.02                                                                                                                                                                                                                                              
report_inches: false                                                                                                                                                                                                                                                     

stepping:                                                                                                                                                                                                                                                                
  engine: I2SO_STREAM                                                                                                                                                                                                                                                    
  idle_ms: 255                                                                                                                                                                                                                                                           
  dir_delay_us: 1                                                                                                                                                                                                                                                        
  pulse_us: 10                                                                                                                                                                                                                                                           
  disable_delay_us: 0                                                                                                                                                                                                                                                    

axes:                                                                                                                                                                                                                                                                    
  shared_stepper_disable_pin: NO_PIN                                                                                                                                                                                                                                     

  x:                                                                                                                                                                                                                                                                     
[... 85 lines cut ...]
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN

#Huanyang:
#  uart:
#    txd_pin: gpio.17
#    rxd_pin: gpio.16
#    rts_pin: gpio.4
#    baud: 9600
#    mode: 8N1
#  modbus_id: 1
#  tool_num: 0
#  speed_map: 0=0% 0=25% 6000=25% 24000=100%

#probe:
#  pin: gpio.27:high
#  check_mode_start: true
ok

I went back a few FluidNC releases just in case it was a recent issue, but they all do the same thing. I doubt a problem this big would have gone unnoticed, so I assume there's something wrong with my configuration. Can you help?

Thanks.

bdring, 2022-05-07:
The reboot is intentional to allow you to fix a bad config file.

Paste in the full config file. What you have here an unreadable mix of formats.

Use a single code block with the yaml tag. If you do not understand that, just attach the file as a .txt text file.

mikey526, 2022-05-07:
[config.yaml.txt](https://github.com/bdring/FluidNC/files/8645840/config.yaml.txt)

bdring, 2022-05-07:
Change I2SO_STREAM to I2S_STREAM

```yaml
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 10
  disable_delay_us: 0
 ```
 
 https://github.com/bdring/FluidNC/wiki/FluidNC-Motor-Setup#engine

mikey526, 2022-05-07:
That did the trick, thank you for your help!

closed 2022-05-07 by mikey526