bdring/FluidNC #410: can fluidNC  run at esp32-s2?

planevina, 2022-05-10:
I bought a esp32-s3 board , and run the install-fs.bat , it tell me ”This chip is ESP32-S3 not ESP32. Wrong --chip argument?“
I found this in wiki
"The ESP32 must be the dual core 240MHz version (ESP32-S2). Any memory size above 4Meg will work. Currently the pre-compiled version will only use 4Meg. The rest of the memory will not be accessible unless you compile yourself with a custom partition. Ask on discord about this.”

so i bought a new esp32-s2 board, and run  the install-fs.bat , it tell me :

win64\esptool.exe --chip esp32 --baud 921600 dump_mem 0x3ff5a018 4 SecurityFuses.bin
esptool.py v3.3
Found 1 serial ports
Serial port COM5
Connecting....
COM5 failed to connect: This chip is ESP32-S2 not ESP32. Wrong --chip argument?

so ... can fluidNC  run at esp32-s2?
thanks .

planevina, 2022-05-10:
[](url)
[image]
[image]

here it is~

bdring, 2022-05-10:
I am not familiar with that variation. You should contact AI Thinker for support.

We typically use the ESP-WROOM-32 type. [Here is our reference page](https://github.com/bdring/FluidNC/wiki/ESP32-Pin-Reference).

It appears to have eFuses set. Several people have bricked their ESP32s by trying to flash when eFuses are set. Our install scripts will not install on chips with eFuses set.

If you want to skip that safety check, you can either....

1. Compile from source and upload.
2. Edit our install scripts to remove the security check.

Do this at your own risk.

Here is a link to our Discord server if you want to discuss this with our group.

https://discord.gg/Ryw9Nbsb

closed 2022-05-10 by bdring