bdring/FluidNC #412: Any change in Config Items(WebUI) -> Initialization error at axes

TorAllex, 2022-05-10:
**FluidNC v3.4.4**
Hello!
First of all, thank for your project, it cool.
I'm a newby, make my first CNC lathe now. All things are work as it should, but couple trouble for me:
1) Can't connect to local WiFi, its password is empty.
2) When I try to change any parameter at WebUI/Config Items, errors occoured:
```
[MSG:ERR: Initialization error at /axes/x/motor0/tmc_2130: For daisy-chained TMC, cs_pin: pin must be configured only once]
[MSG:ERR: Initialization error at /axes/y/motor0/tmc_2130: spi_index: must be unique among all daisy-chained TMCs]
[MSG:ERR: Initialization error at /axes/z/motor0/tmc_2130: spi_index: must be unique among all daisy-chained TMCs]
[MSG:ERR: Initialization error at /axes/a/motor0/tmc_2130: spi_index: must be unique among all daisy-chained TMCs]
```
Boot Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
```
Startup Messages:
```
[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:TMC2130_Rel_testA.yaml]
[MSG:WARN: max_rate_mm_per_min value 180000.000 constrained to range (0.001,100000.000)]
[MSG:INFO: Machine 4 Axis SPI stealthChop +Relay]
[MSG:INFO: Board 4 Axis SPI Daisy]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:254ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.12 Dir:gpio.14 CS:gpio.17 Disable:NO_PIN Index:1 R:0.110]
[MSG:INFO: Axis Y (150.000,210.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.27 Dir:gpio.26 CS:NO_PIN Disable:NO_PIN Index:2 R:0.110]
[MSG:INFO: Axis Z (150.000,450.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.15 Dir:gpio.2 CS:NO_PIN Disable:NO_PIN Index:3 R:0.110]
[MSG:INFO: Axis A (150.000,510.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2130 Step:gpio.33 Dir:gpio.32 CS:NO_PIN Disable:NO_PIN Index:4 R:0.110]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:gpio.4 Out:gpio.25 Dir:NO_PIN Freq:250Hz Res:18bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: Connecting to STA SSID:OpenWrt]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connection failed]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]
```
[TMC2130_Rel_testA.yaml.zip](https://github.com/bdring/FluidNC/files/8664541/TMC2130_Rel_testA.yaml.zip)

bdring, 2022-05-10:
I have never tried to connect to a wifi with no password. Did you send **$Sta/Password=**

We became aware of those errors last week. It occurs when you try to change a stepper driver parameter with daisy chained SPI drivers. The error is benign and you can ignore it. We will suppress the the error on the next release.

TorAllex, 2022-05-10:
Thanks for quick reply!
> **$Sta/Password=**
Of course, I've try it
```
$Sta/Password=
[MSG:ERR: Setting length error]
ERROR:2
Missing the expected G-code word value or numeric value format is not valid.
```
> The error is benign and you can ignore it.
But have "**ERROR:8**"
```
$J=G91 G21 F100 Y1
ok
<Jog|MPos:0.000,0.048,0.000,0.000|FS:100,0|Ov:100,100,100>
[MSG:ERR: Initialization error at /axes/x/motor0/tmc_2130: For daisy-chained TMC, cs_pin: pin must be configured only once]
[MSG:ERR: Initialization error at /axes/y/motor0/tmc_2130: spi_index: must be unique among all daisy-chained TMCs]
[MSG:ERR: Initialization error at /axes/z/motor0/tmc_2130: spi_index: must be unique among all daisy-chained TMCs]
[MSG:ERR: Initialization error at /axes/a/motor0/tmc_2130: spi_index: must be unique among all daisy-chained TMCs]
$J=G91 G21 F100 Y-1
ERROR:8
Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job.
<Alarm|MPos:0.000,1.000,0.000,0.000|FS:0,0>
```

bdring, 2022-05-10:
I tried **$STA/Password=** and got a length error. The minimum length of a password is 8.

It is controlled by this line `static const int MIN_PASSWORD_LENGTH = 8;`

You could try changing that to 0 and compiling yourself. I have no desire to test this with my router :-)

TorAllex, 2022-05-10:
Ok, I'll try to compille.
What should I do with "ERROR:8"? It stopped responce for $ commands

bdring, 2022-05-10:
You can get the description of any error with **$E=\<number\>** or get the list of all errors with **$E**

```
$e=8
8: Command requires idle state
```

The command you are trying to send requires you to be in Idle mode. $H or $X will do it at startup.

The same is true for alarms **$A=\<number\>** or $A for them all.

TorAllex, 2022-05-11:
> You could try changing that to 0 and compiling yourself

Ок 
```
[MSG:INFO: Connecting to STA SSID:OpenWrt]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.134]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://TaigGRBL.local/]
```

TorAllex, 2022-05-11:
```
[MSG:INFO: '$H'|'$X' to unlock]
$X
[MSG:INFO: Caution: Unlocked]
ok
$/axes/y/acceleration_mm_per_sec2
$/axes/y/acceleration_mm_per_sec2=5.000
ok
$/axes/y/acceleration_mm_per_sec2=15
[MSG:ERR: Initialization error at /axes/x/motor0/tmc_2130: For daisy-chained TMC, cs_pin: pin must be configured only once]
[MSG:ERR: Initialization error at /axes/y/motor0/tmc_2130: spi_index: must be unique among all daisy-chained TMCs]
[MSG:ERR: Initialization error at /axes/z/motor0/tmc_2130: spi_index: must be unique among all daisy-chained TMCs]
[MSG:ERR: Initialization error at /axes/a/motor0/tmc_2130: spi_index: must be unique among all daisy-chained TMCs]
ok
$X
error:152
```

TorAllex, 2022-05-23:
Ok, error fixed on v3.4.5
$MI still require for apply changes.

MitchBradley, 2022-05-23:
We require $MI to commit the changes because changing the individual items one at a time could lead to inconsistent setups.

closed 2022-05-23 by MitchBradley