bdring/FluidNC #414: Increase Spinup/Spindown constaint values

daxliniere, 2022-05-11:
My VFD and spindle takes just over 30 seconds to spin up to 24000 rpm, but FNC has a maximum ceiling of 20 secs.

[MSG:WARN: spinup_ms value 31000 constrained to range (0,20000)]
[MSG:WARN: spindown_ms value 31000 constrained to range (0,20000)]

bdring, 2022-05-11:
WOW! My oatmeal gets up to speed faster every morning.

Is that the timing you want? What spindle class are you using?

daxliniere, 2022-05-11:
hahaha
It's a 1.5kW 3-phase spindle running with 3kW VFD. I couldn't tell you a class (nothing mentioned), but it's fairly powerful.

daxliniere, 2022-05-11:
Ahh. It has external input controls for 0-5V, 0-10V and 20mA, though I have no idea how the last one works.
I wish it was RS485, but that tripled the price, which didn't seem worth it.

This is the model: https://www.vevor.co.uk/variable-frequency-drives-c_10745/4hp-3kw-variable-frequency-drive-inverter-vfd-13a-220v-motor-speed-control-vsd-p_010692009268

MitchBradley, 2022-05-11:
Please tell us everything about your setup - controller board, interface electronics that you use to control the VFD, etc.  This will take forever if we have to keep asking for information.

bdring, 2022-05-11:
I think it is a rebrand huanyang.

PD014 is the accel time. Try changing it to something reasonable like 8 seconds.

daxliniere, 2022-05-11:
Hi Mitch. Spin-up/-down values are independent of those other factors in this particular situation.

The testing routine was setting the VFD to maximum speed, then pressing the enable button at the same time as the stopwatch, then waiting for it to hit the target speed.

daxliniere, 2022-05-11:
Yeah, rebranded Huanyang.
30 secs seems like a nice slow warm-up, which the manual for the spindle recommends.

bdring, 2022-05-11:
Typically a spindle has a warmup sequence. This allows the grease to evenly distribute before you use it. 

You typically go to several speeds and dwell at that speed for several minutes.

12000 for 3 minutes
15000 for 3 minutes
24000 for 3 minutes

Then you can start and stop the spindle quickly as long as it does not sit too long...like 30 minutes.

Using a 31 second spinup will drive you crazy in production.

Most people with DIY machines and cheap spindles ignore it anyway.

daxliniere, 2022-05-11:
Awesome, thanks so much for the info, Bart. I really appreciate you taking the time to explain it.

I'm certainly happy to have a shorter spin-up time! I'll set that when I'm next at the workshop.

All the best!
Dax.

closed 2022-05-11 by daxliniere

MitchBradley, 2022-05-11:
> Hi Mitch. Spin-up/-down values are independent of those other factors in this particular situation.

Not necessarily.  We have some soft-start code kicking around that depends on the particular interface.  I wanted to check to see if it would interact badly with your setup, which is why I want to know the details.