bdring/FluidNC #415: Error setting macro from web interface

wu5, 2022-05-11:
Hi,

When I try to change a macro from the web interface configuration section, I get an error:

_Set failed
Error 500 :Error: Invalid value for setting_

The error occurs, as far as I can see, only if the macro line contains the command to run the file from the SD card, like `$SD/Run=test.nc`.

If I set up the macro via the terminal by running the command `$/macros/macro0=$SD/Run=test.nc`, no error occurs. When setting other commands, like `$Home`, the error will not be thrown either.

Please only submit bugs for the **latest** version of the code. You can check the version number in the startup messages.

**Do not add the bug label**. The developers of this repo will do that after confirming the issue.

Please provide the [FluidNC boot messages per this page](https://github.com/bdring/FluidNC/wiki/Requesting-Help#fluidnc-boot-messages).

Please answer the following questions.

What version of the firmware are you using?

Is the problem repeatable?

Under what conditions does the bug occur?

**Important** If you paste firmware code, please use [Markdown Code and Syntax Highlighting](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code) with language C++. Use the three back tick method.

```C++
  #define EASIER_TO_READ true
```

MitchBradley, 2022-05-11:
I found a way to fix it without too much trouble.  The fix will probably be in the next release.

Please note that, if you change a configuration setting of this type, the change will not persist across resets unless you save the config file.

wu5, 2022-05-12:
Thanks.

The news about the development of a new version of WebUI is great.

MitchBradley, 2022-05-12:
> The news about the development of a new version of WebUI is great.

Please note that WebUI is controlled by a developer who is not part of this project.  People often ask us for WebUI improvements that strain our ability to work on FluidNC features and also coordinate with the separate project.

commit 1ba4f6a4 (2022-05-12): Merge pull request #416 from bdring/EqualInSetting

Fixed #415 - equals character in setting value fails with WebUI
commit a00c3a87 (2022-05-12): Fixed #415 - equals character in setting value fails with WebUI