bdring/FluidNC #417: M4 Laser Command Failed in Lightburn

TinkerBoi, 2022-05-12:
I'm using the latest version of LightBurn 1.1.0.4. I'm doing a bench test without an actual laser hooked up or enabled on the power supply for safety reason. When I set the laser to GRBL-M3 it worked but with GRBL selected that run M4 for pwm power control it returned the following error and immediately stopped. Line 2 is the M4 command, if I remove this line and run the gcode file as a cut file, it worked fine. I'm not sure if my config.yaml for the laser is properly setup. This is a custom CO2 laser machine. Any help is greatly appreciated!

Starting stream
Layer C00
error:20
Unsupported or invalid g-code command found in block.
On or near line 2:
Job halted
Stream completed in 0:00

MitchBradley, 2022-05-12:
Please supply your config file and the startup messages.

TinkerBoi, 2022-05-12:
I forgot to mention that I'm running the Espduino-32 board with the cnc shield V3 two axises on x and y. So far I got everything to work except the above issue and also after finishing a cut (in GRBL-M3 machine profile) it crashed into Y end switch before the X reached the finish position (but that is another issue). 

I attached the config file and here is the startup messages:

[config.txt](https://github.com/bdring/FluidNC/files/8679257/config.txt)]

[MSG:INFO: FluidNC v3.4.4]
[MSG:INFO: Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:INFO: Configuration file:config.yaml]
[MSG:WARN: Ignored key pulloff_mm]
[MSG:WARN: Ignored key pulloff_mm]
[MSG:INFO: Machine MACHINE_ESPDUINO_32]
[MSG:INFO: Board ESPDUINO_32]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.12]
[MSG:INFO: Axis X (0.000,600.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.13:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.25 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.5:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.19 Dir:NO_PIN Freq:5000Hz Res:13bits]
[MSG:INFO: Using spindle PWM]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC-ESP32 IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]

bdring, 2022-05-12:
M4 has 2 uses.
- With spindles it is used for reverse direction
- With lasers it is used for a special laser power mode.

You have a spindle defined, not a laser. FluidNC sees that you do not have a direction pin in the config file for the spindle so it is not allowing the M4 command.

Change to using a laser.

https://github.com/bdring/FluidNC/wiki/Fluid-Spindle-Setup#laser

TinkerBoi, 2022-05-12:
> M4 has 2 uses.
>
> * With spindles it is used for reverse direction
> * With lasers it is used for a special laser power mode.
>
> You have a spindle defined, not a laser. FluidNC sees that you do not have [...]

Thanks so much Bart! I knew it was something simple I overlooked! Also is it correct to set the max power mapping at 1000=100%? I assumed this should match the S-value max in Lightburn.

Btw, I wanted to say thank you for creating such an amazing framework with FluidNC. I was frustrated at first due to the overwhelming amount of info to learn and convert from GRBL 1.1 but im glad I did. Now this seems very similar to Marlin firmware in 3D printing.

bdring, 2022-05-12:
You are correct with the S value. Most people use 255 or 1000.

TinkerBoi, 2022-05-12:
> You are correct with the S value. Most people use 255 or 1000.

Thank you! Working great now!