bdring/FluidNC #418: Finish Position Crashed in Lightburn with Alarm 1

TinkerBoi, 2022-05-12:
When finishing a cut, the laser tried to return to the finish position (in Lightburn I jog the laser to +10mm on x and y to clear the end switches and hit the set finish position), but instead it didn't stop and trigger the end-stop (usually on Y) and halt. Working dimensions: X=600, Y=300. Is my config file correctly set for these axises?

Starting stream
Layer C00
ALARM:1
Hard limit triggered. Machine position is likely lost due to sudden and immediate halt. Re-homing is highly recommended.

[config.txt](https://github.com/bdring/FluidNC/files/8679534/config.txt)

Starting Messages

MSG:INFO: Machine MACHINE_ESPDUINO_32]
[MSG:INFO: Board ESPDUINO_32]
[MSG:INFO: SPI not defined]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable gpio.12]
[MSG:INFO: Axis X (0.000,600.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.16 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.13:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.25 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:     All Limit gpio.5:low]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Laser Spindle Ena:NO_PIN Out:gpio.19 Freq:5000Hz Res:13bits Laser mode:On]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC-ESP32 IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
Grbl 3.4 [FluidNC v3.4.4 (wifi) '$' for help]

TinkerBoi, 2022-05-12:
> It is hard to know what happened with the data provided.
> 
> Did you use a jog command to move or a gcode command? Please give the actual command.
> 
> Gcode moves in work coordinates, so it is possible you have an offset set. Send $# to see the offsets.
> 
> If you don't understand offsets see this video.
> 
> https://www.youtube.com/watch?v=fGtbkVJBXyE

I used the arrow keys to move the head to the finish position (after homing) so it is at x10, y10.

When I send $# this is returned:

$#
[G54:30.000,-17.000,0.000]
[G55:0.000,0.000,0.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
ok

TinkerBoi, 2022-05-12:
Also I can't find any information on what the deactivate_parking feature (or rather parking) does. I'm looking for a way to auto home after it finished a job, is this it?

bdring, 2022-05-12:
Parking is a standard Grbl feature that is usually associated with a door switch.

Search for Grbl Parking

MitchBradley, 2022-05-12:
Parking is used in conjunction with the safety door feature - if the safety door is opened, the spindle will automatically move to a specified Z coordinate.  It is also invoked by M30 - the program end gcode that is sometimes used at the end of a GCode program.  Parking is probably not helpful in your case because you don't have a Z axis, and also because we hardly ever test it so I have no idea whether it works right.

There is a G29 command that will return to a reference position, and a corresponding G29.1 to set the reference point.  That might do what you want if you can manage to get it included at the end of the gcode program you are running.  But if you can do that, you could also just put a G1 command to go where you want.

"At the end of a job" is not as specific as you might think.  "Job" generally means "execute a sequence of GCode commands, perhaps from a file, perhaps streamed from a sender".  What is in that sequence is outside the control of FluidNC, so to control what happens at the end requires knowledge of the software that generates the program and/or the sender that streams it.  If Lightburn is generating your GCode, you need to interact with it to do special things at the end - so you need to ask the question on a Lightburn forum.

closed 2022-05-12 by MitchBradley